nightscout/AndroidAPS #1966: Wear app: Double tapping the "Digital Style" watchface graph area no longer changes graph resolution

vanelsberg, 2022-07-25:
**AndroidAPS**: 3.1.0.2-dev-a
**Build**: 0599E81
**Flavor**: Fullrelease, developer mode

**Problem:**
Double tapping the Wear App "Digital Style" watchface area no longer changes graph resolution.

*Additional info*:

- Running AAPS and Wear app from same build (latest dev)
- Graph resolution can be change through settings as usual

rICTx-T1D, 2022-07-25:
Tap double on time digests.

vanelsberg, 2022-07-26:
> Please try to tap double on time digests. Sometimes you need one or two additional taps to activate your interactive mode from wearOS. My Fossil Carlyle need 4 taps to switch first time and double tap to switch to next.

1. On Watch 4 tapping work reliable always. 2 taps just (used to) works
2. Tap area is/was graph area
3. Resolution does not change by any tapping

rICTx-T1D, 2022-07-26:
When I tap multiple times on weekday name, the scale is changing.  Yes the tap area is moved incorrectly. 

Can you confirm the scale-change?

vanelsberg, 2022-07-26:
Nope!

vanelsberg, 2022-07-27:
> When I tap multiple times on weekday name, the scale is changing...
> ... the **tap area is moved** incorrectly.

(I checked again) Yes - you are absolutely right!

rICTx-T1D, 2022-07-27:
@vanelsberg Can you modify the headline of this issue? Tapping-Zone is incorrect placed.

vanelsberg, 2022-07-27:
I'd rather leave it like it is. Because that is the actual UI experience (caused by "Tapping-Zone is incorrect placed")

Analyzing actual problem and solving is to the devs to decide.
Comments here do point in that direction so solving should not be a problem ;-)

PR #2026 (2022-08-18): fix: wear digital chart zoom
Make watch face Digital Style zoom by tapping on the full right column, including the chart. Fixes #1966
closed 2022-08-18 by MilosKozak