nightscout/AndroidAPS #1973: [3.1.0.2] problems on small displays with new abbreviation for „days“ and „hours“

fezulin, 2022-07-29:
AndroidAPS 3.1.0.2 (compared in pictures with AAPS 3.0.0)
Used mobile: Pixel 4, Display 2280 x 1080 Pixel

Using a language specific abbreviation for „days“ and „hours“ seems to cause trouble in GUI on mobiles with small displays, if the new abbreviation has more then one letter. 

Picture 1: If pump is disconnected for 2 hours (TWO hours!), in 3.1.0.2 the GUI shows just 1(!) hour, minutes cannot be displayed. 

Picture 2: The „U“ for „untis“ cannot be displayed. I agree: It is not hard to guess, what is meant. ;o) 

Picture 1: 
[image]

Picture 2:
[image]

maheini, 2022-07-31:
Same here. Mostly a problem caused by the translation, which isn't really a short version in German. I don't have a very small screen (1080 x 2160 wide) but still faced the missing numbers.

OFlannigan, 2022-07-31:
Can also confirm, though not on a small screen. See also issue #1960. For short term solving, I switched the language to english.

fezulin, 2022-07-31:
So this issue (#1973) might be closed, as it is a duplicate to #1960.
Sorry - I did not see #1960, when I posted this issue.

closed 2022-07-31 by fezulin