nightscout/AndroidAPS #1975: Unstable connection (totally unuseable) V3.1.0.2 Dana i (Samsung S20FE and Xiaomi Poco X3 Pro)

MowgliChild, 2022-07-29:
Hi

Issue requested by @MilosKozak 

Time: Ive just moved from (V3.0 with VP) to (3.1.0.2 with  Dana i)

Pump has intermittent comms causing a variety of (catestrophic) failures:

No treatment buttons
Poor BG control (Im bolusing in fasting). Not sure what the pump is doing at all.  
"Large time difference"
"Application needs BT permission"
"No profile set"
"Pump not initialized, profile not set!"
"Failed to upload basal profile"

I've been trying to connect for 5 days. My control is terrible, I feel bad. I have Dana i, samsung S20FE A12 (and a Xiaomi Poco X3 Pro) FSL1 Bubble XDrip, I'm able to import and advance my VP objectives, have the latest build 3*1*0*2

I've had all sorts of errors  time miss match ( transpires this is a connection issue), pump not initialized, profile not set, failed to load basal profile etc. The errors are symptoms, not cause and I'm not a mechanic or doctor. Most of us do this only once so don't have a great experience at diagnosing aaps.

These errors are repeated until I have to switch off the volume (removing my BG alarm) or delete AAPS.
I have been in contact with Advanced Therapeutics who had me run some tests on AnyDana and it responds fine. I have also sent them my apk. The pump was also working fine in Camaps which I was using previously with no problems. They are passing this on to SOOIL.
Yes I've been careful with pairing management.

Ive read all the related posts by searching the group.

My phones are not strange outliers.

My LP is complete and sent.

Ive attached the log files that were present today.

--------------
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/9220597/AndroidAPS.log)
[AndroidAPS._2022-07-28_19-14-00_.0.zip](https://github.com/nightscout/AndroidAPS/files/9220598/AndroidAPS._2022-07-28_19-14-00_.0.zip)
[AndroidAPS._2022-07-27_09-44-49_.0.zip](https://github.com/nightscout/AndroidAPS/files/9220599/AndroidAPS._2022-07-27_09-44-49_.0.zip)
[AndroidAPS._2022-07-26_22-58-42_.0.zip](https://github.com/nightscout/AndroidAPS/files/9220600/AndroidAPS._2022-07-26_22-58-42_.0.zip)
[AndroidAPS._2022-07-25_09-02-08_.1.zip](https://github.com/nightscout/AndroidAPS/files/9220601/AndroidAPS._2022-07-25_09-02-08_.1.zip)
[AndroidAPS._2022-07-25_09-02-08_.0.zip](https://github.com/nightscout/AndroidAPS/files/9220602/AndroidAPS._2022-07-25_09-02-08_.0.zip)
[AndroidAPS._2022-07-24_13-08-28_.1.zip](https://github.com/nightscout/AndroidAPS/files/9220603/AndroidAPS._2022-07-24_13-08-28_.1.zip)
[AndroidAPS._2022-07-24_13-08-28_.0.zip](https://github.com/nightscout/AndroidAPS/files/9220605/AndroidAPS._2022-07-24_13-08-28_.0.zip)

MowgliChild, 2022-07-30:
Hi people,

Log files above for Samsung S20FE
Log files below for Xiaomi Poco X3 Pro

Please let me know if I can be of any other assistence. 

Thanks

[AndroidAPS._2022-07-25_03-42-01_.0.zip](https://github.com/nightscout/AndroidAPS/files/9226687/AndroidAPS._2022-07-25_03-42-01_.0.zip)
[AndroidAPS._2022-07-26_20-07-32_.0.zip](https://github.com/nightscout/AndroidAPS/files/9226688/AndroidAPS._2022-07-26_20-07-32_.0.zip)
[AndroidAPS._2022-07-28_23-58-23_.0.zip](https://github.com/nightscout/AndroidAPS/files/9226689/AndroidAPS._2022-07-28_23-58-23_.0.zip)
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/9226690/AndroidAPS.log)

MowgliChild, 2022-08-01:
Ok, I have rhe solution and you can close this issue.

This is specific to Dana (im on I and can't comment about RS). 

Enter doc mode (video on YouTube with Advanced Therapeutics) and make sure UTC is correct

MowgliChild, 2022-08-01:
There are many symptoms/ errors which make diagnosis of the problem very difficult.

closed 2022-08-01 by MowgliChild

MilosKozak, 2022-08-01:
wasn't there a message that setting time failed?

MowgliChild, 2022-08-16:
On Dana i there are 2 UTC settings, one in patient access to pump settings and the other in doctor mode. 
Adjusting UTC on the pump resolved many issues which are listed above in the original ticket, one of which is time difference.