nightscout/AndroidAPS #1978: [3.1.0.2] Still issues with Omnipod Eros pods / reference to #1627

jalbers01, 2022-07-31:
Referring to #1627, unfortunately my previous issue seems not to be resolved entirely. After I was not able to work with AAPS version 3.0.0.1, I switched back to 2.8.2.1 and was absolutely satisfied with this older version, it simply worked. Now as I am forced to update again (BTW: I would be happy to continue using the older version, but why are 2.8.2.1 users not allowed to keep the running system alive?), I tried to use the freshly compiled version 3.1.0.2.
I have this issue: After importing my JSON file, erroneously an Eros pod is shown to be active. As it still cannot be deactivated, I can’t start a new one. Interestingly, the same JSON file (and I tried several ones) works with 2.8.2.1 without any error. How can that be explained, I what could I do to start the system with a new pod? Thank you.
Attached: Screenshot of 3.1.0.2 with pod active after importing the JSON file
[image]

happygirlnyc70, 2022-08-03:
@jalbers01 in the picture your rileylink is not connected. Does it ever connect? If it does connect is the option to discard pod available?

jalbers01, 2022-08-06:
I had some issues with connecting to Riley. I assume something is wrong with my json file, because with an older version of the json file I was able to connect to Riley and to start a new pod. Looks weird to me, but just now it's up and running with 3.1.0.2. However, I had to activate the profile after changing to the virtual pump driver temporarily and then back to Omnipod to activate the new pod. Very much different and much more complex than it was with 2.8.2.1

happygirlnyc70, 2022-08-06:
There was another issue thecubic reported.  He also had to use the virtual pump work around.  Milos just released 3.1.0.3. I plan to install it at my pod change today. I noticed there is a fix for the issue so everything should work just fine now.  @vanelsberg did great work fixing the original problem with pod errors. Hopefully your problem is solved now.

vanelsberg, 2022-08-07:
> I what could I do to start the system with a new pod?

Due to a change in the way Pod state is handled, upgrading from 2.8 with an active Pod may give problems. To solve your problem:

1. You can try the work around as described in the original issue #1963: Set the pump driver to the virtual pump, which allowed the profile switch to activate, then switched pump drivers back to Omnipod Eros and retry deactivating.
3. Temporarily switch to developer mode. It offers an additional button to "force-remove" current Pod registration.

See also:
https://github.com/nightscout/AndroidAPS/issues/1963

closed 2022-10-04 by MilosKozak