nightscout/AndroidAPS #1988: App freezing on 3.1.0.2

yhonatanb20, 2022-08-04:
Had an app freezing on 3.1.0.2, thought it should have been fixed: when tapping on aaps icon, got white screen with aaps logo for 5 seconds, than this screen closed, on second tap app lunched and "aaps started" note now displayed on graph
[image]

Andries-Smit, 2022-08-04:
@yhonatanb20 thanks for reporting the issue. Could you please add the log files so the issue could be investigated.

yhonatanb20, 2022-08-04:
[AndroidAPS._2022-08-04_21-11-54_.1.zip](https://github.com/nightscout/AndroidAPS/files/9262709/AndroidAPS._2022-08-04_21-11-54_.1.zip)
[AndroidAPS._2022-08-04_21-11-54_.0.zip](https://github.com/nightscout/AndroidAPS/files/9262710/AndroidAPS._2022-08-04_21-11-54_.0.zip)
Here are all the files under the directory aaps/logs/info.nightscout.androidaps with date and time corresponding for the issue. Hope that's what you meant?

MilosKozak, 2022-08-14:
it's impossible to find anything in logs without exact knowledge what to look for. When this happen again open new isse with time, logs etc, exactly like described in issue template

closed 2022-08-14 by MilosKozak