nightscout/AndroidAPS #1995: Autosens freezing / failing to update - 3.1.0.2

BeFlatLine, 2022-08-07:
Version 3.1.0.2.

Time of issue: 12:36 PM roughly. 

Desceiption: Autosens appears to freeze its update sporadically since updating from 3.0 to 3.1.0.2.  The Grey dot of starting AndroidAPS is from me troubleshooting the issue when it occurred previously. 

NOTE: I changed profiles to a lesser profile because I've been more sensitive than normal to insukin lately. My body has swings where I'll need 10 - 20% more insulin for a day or two, but the past 2 days have been the opposite (where I've needed less insulin than usual). Not sure if that is playing an impact. 

[image]
[image]
[image]
[image]
[image]

BeFlatLine, 2022-08-07:
[AndroidAPS_LOG_1659891342672.log.zip](https://github.com/nightscout/AndroidAPS/files/9277193/AndroidAPS_LOG_1659891342672.log.zip)
Adding the log file.

BeFlatLine, 2022-08-12:
Updated to 3.1.0.3 and issue is still occurring. 
[image]

MilosKozak, 2023-01-21:
improved by caching last autosens value in dev already

closed 2023-01-21 by MilosKozak