nightscout/AndroidAPS #1997: Enable the nsclient local broadcast even when uploads are disabled

4RK4N, 2022-08-08:
Time: always
Build: 3.1.0.2
Steps to reproduce:
1. setp nsclient to only upload on wifi when charging or disable all nsclient uploads manually via the sync settings
2. set xdrip+ to recieve treatments from local broadcast, disable treatments download from ns
3. enable aaps nsclient local broadcast to xdrip+
4. enter any treatment (bolus, profile change, sensor change, etc)
5. see the treatmwnt in aapa treatments logs
6. see no treatment in xdrip
7. re-enable nsclient uploads
8. wait for upload queue to be cleared
9. see treatment in xdrip+ now

Apparently the local broadcast of nsclient ssems to be tied to the upload queue, specifically getting sdnd after the queue is cleared. if no uploads happen, so if the queue doesnt get cleared, no local broadcast is sent either.
that kinda contradicts the term "local" f it only gets sent when im "online".

proposed solution: separate the queue uploads and the local broadcast. local broadcasts shouldnt be tied to a "be online" condition.

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2022-08-08:
I already explained it elsewhere
This function is not xDtip - AAPS synchronization but broadcast of data coming from NS to avoid multiple downloads.
So it's working like expected

closed 2022-08-08 by MilosKozak

nlof, 2022-08-29:
@MilosKozak  With the looming shutdown of Heroku's Nightscout service, there should be a way for Xdrip to receive treatments from AAPS directly, without involving NS at all...

I don't think It makes sense to force people to use NS and the internet in order for the local Xdrip on the same phone to receive AAPS treatments...