nightscout/AndroidAPS #1998: SMS delayed delivery by 2 hours

LadyLadedah, 2022-08-09:
today (09.08.2022) I tried to SMS bolus my daughter 0.10 at lunch (approx. around 12.30pm). The bolus did not deliver.
Later in the afternoon when I tried to bolus 0.75 (15:00) the SMS delivered the missing 0.10 instead. The 0.75 has not been delivered.
Confirmation of 0.1 delivery was received in treatments and on N/Scout
This is a concern.
Propose we have a function to cancel or override a SMS Command if its clear SMS Command has not performed within 30 mins of sending?

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2022-08-10:
without logs it's impossible to help

LadyLadedah, 2022-08-10:
Apologies I’ve appalling IT knowledge

When you say logs what are you looking for?

Happy to supply!

MilosKozak, 2022-08-10:
https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html?highlight=log%20files
be sure you are uploading all logs from mentioned period

koelewij, 2022-08-12:
Also take into account, that there can be many factors that affect the delivery time of sms messages. See for instance https://messente.com/blog/sms-delay

andyrozman, 2022-08-12:
SMS delays can be due to network issues (especially if persons in "talk" are on different networks/providers), and even on same provider sometimes it can happen that messages are delayed... SMS is not like an instant messager, where you know that message was received...

LadyLadedah, 2022-08-12:
Thanks Andy but we are on the same network (vodafone), and I had been SMS
bolusing a number of times that morning.

I recall it was going fine. But I'll update the logs over the weekend.

koelewij, 2022-08-12:
Don't wait too long, Logs usually report up to 3 days as far as i know.

LadyLadedah, 2022-08-16:
Hello I downloaded the log last friday. Before I hand it over, am I sharing
any sensitive data?

closed 2022-10-04 by MilosKozak