nightscout/AndroidAPS #2001: Possibility to enter insulin keys instead of IC-factor in LP

GerritosSteen, 2022-08-10:
Is there a possibility to enter the insulin keys instead of the IC-factor in the locale profile? I might be the only one but I always get confused when changing the insulin rates for bolus for my 7yo son. I have been diabetic type 1 by myself for the past 23 years and have always counted with the insulin keys. With the IC I always need to think twice in order to correctly enter the new value.

It would be great to have the possibility to choose whether to use the IC-values or the insulin keys for entering in the local profile.

IC: grams of carbs that will be covered by 1 unit insulin (i.e. 6,7)
insulin key: carb factor in unit insulin / carbs (i.e. 1,5)

MilosKozak, 2022-08-10:
I believe this option would be more confusing because most of the world and all other loop systems are using IC. It could lead to many  mistakes.
So no for now.
Btw where are you from?

GerritosSteen, 2022-08-10:
What a pity :-(
Rest follows by PN in FB ;-)

closed 2022-08-10 by GerritosSteen