nightscout/AndroidAPS #2002: Last 3103.devA kills connection to DASH

robertrub, 2022-08-10:
@MilosKozak The latest dev build killed things. For DASH, no more pod history (shows a blank page and then slows the AAPS UI). No more beep the pod, nor connect to pod.
[image]

Reinstalling the older devA gets back the pod connection but losts selected profile, BG history etc.

Here is the log. Look around and before 13:55

[image]

robertrub, 2022-08-10:
Help !!! I downgraded to 

[image]

but COB stays at 0 whatever I do (enter carbs via wizard or manually)

[image]

MilosKozak, 2022-08-10:
it seems db update is not handled properly ..... i'll check on my phone

elgrupo123, 2022-08-10:
Here is a similar problem. The connection to the Dash is there, but no bolus delivery possible
[image]
[image]
.

MilosKozak, 2022-08-10:
unable to replicate this issue.
But PR 1999 is removed now

robertrub, 2022-08-11:
I updated from this dev

[image]

to this dev

[image]

I don't know if it is important or not.

philipgo, 2022-08-11:
Same problem here, Pod is connected but Basal cannot be set.

I received this message when trying to connect a new pod:[image]

Philoul, 2022-08-12:
@MilosKozak Unfortunatly I have updated my PR #529 with dev branch including #1999 to resolve conflicts with insight Driver in kt and Notes update...
What's the best way to fix it ? close PR and make a new one (I don't know how to make a hard reset (on commit) on a remote branch...) ?

robertrub, 2022-08-12:
@MilosKozak VERY strange. After the incident, I installed an older 3103dev A then installed MASTER 3103 and have been using master since. Since 0h today, I got my stats partially back (see number of days  insulin stats vs BG stats) 

[image]

But the BG stats for the last 2 days are wrong, I was definitely NOT 100% yesterday...)

[image]

robertrub, 2022-08-14:
Seems that the stats got solved by themselves.

[image]

closed 2022-08-14 by MilosKozak