nightscout/AndroidAPS #2003: [3.1.0.3] "Closed loop" button disappears after being on Open loop

nlof, 2022-08-10:
When trying to close loop (Omnipod) after being on open loop temporarily, the "closed loop" button disappears from the menu. I need to switch to "low BG suspend" first and only after that does "closed loop' button show up. 
Sequential screenshots as well as log attached. It occured multiple times, the last @ 19:09 in the logs.

happygirlnyc70, 2022-08-17:
I can confirm this behavior for both dev and master version. I can attach logs the next time it happens.

Andries-Smit, 2022-08-17:
Its a bug, 
You can only close the loop if objective 6 is completed.
https://github.com/nightscout/AndroidAPS/blob/master/app/src/main/java/info/nightscout/androidaps/dialogs/LoopDialog.kt#L156
To complete objective 6 you have to close it...
https://github.com/nightscout/AndroidAPS/blob/master/app/src/main/java/info/nightscout/androidaps/plugins/constraints/objectives/objectives/Objective6.kt#L21
Caused by commit https://github.com/nightscout/AndroidAPS/commit/c211c2e4c3abf1b71557751e58b02088cbe062e1
@MilosKozak what is there alternative, to prevent deadlock and keep the function?

happygirlnyc70, 2022-08-17:
To be clear, you are saying this is a bug. I have completed all of the objectives.

PR #2025 (2022-08-18): Fix: closed loop button visbility
The closed Loop button should be available when objective 6 has been accomplished in the past. Fixes #2003
(Completed does do an active check if the current conditions are completed, which will never be, as it requires the state to closed)
closed 2022-08-18 by MilosKozak