nightscout/AndroidAPS #2009: 3.1.0.3: profile shift fails

justmara, 2022-08-11:
this is result of profile shift `+1h` for `2h30m`
[image]

this is profile shift `-1h`
[image]

justmara, 2022-08-11:
updated to 3.1.0.3 and it works fine, sorry :)

justmara, 2022-08-15:
Re-tested only to find than it fails when switched from nsclient.
This is how it is set on nsclient 3.1.0.3:
[image]

This is how this event is received from NS by main AAPS
[image]
seen in treatments
[image]
applied in AAPS
[image]

Exactly same settings set from AAPS itself works fine

justmara, 2022-08-15:
Both these events in NS. Seems that it is sent to NS with wrong timing 

[image]

MilosKozak, 2022-08-15:
hopefully fixed by https://github.com/nightscout/AndroidAPS/commit/f7951ec6141b3584790bc88bebe66527a7053b0e

closed 2022-08-15 by MilosKozak