nightscout/AndroidAPS #2010: Enhancement request - New automation triggers CGM sensor insert and Pump site change events

kristinsmotz, 2022-08-12:
I'd like to be able to trigger automations from CGM Sensor insert and Pump site change events. 
Use case: 
After CGM change we often find that the first 1-2 hours give rapidly increasing sensor values, which trigger several SMB's. After 3-4 hours the sensor values drop (significantly more than fingerprick values), so we get alerts for extreme lows even though actual glucose value is around target 5.5-6.5. If I run a Temp Target for 6.7mmol for 4-6 hours after CGM insert we don't get the SMB's during initial too high readings, don't get a sharp drop from the SMB's and don't get the false severe lows. We do most site changes right before bedtime, and around her evening meal, so we tend to have somewhat of a rise around the time readings start coming in, and drop after, but on sensor change days we get unusually high numbers at first, then unusually low numbers in the middle of the night, often hourly from 2-6 AM, so it's affecting sleep.  If I run a Temp Target for 6.7mmol for 4-6 hours after CGM insert we don't get the SMB's during initial too high readings, don't get a sharp drop from the SMB's and don't get the false severe lows. 
I'd like to set an automation to run a temp target after each CGM insert, so that I don't have to remember to put it on manually.  (Example picture: before this we had a high and several SMB's so we assumed the first low was real, treated it, then got several more sharp drops. I did fingerpricks on next low alert and had to make 3 calibrations before sensor stopped giving false lows)
[image]

On Omnipod change days (Pump site change) we have a different problem, where the old site tends to have bad absorbtion/high frequency of tunneling, so we change the site before dosing for evening meal, bolus from new pod and it has delayed absorbtion. We try to do extra long pre-bolus, but often insulin doesn't kick in for 2-3 hours, causing more SMB's to be triggered long after carbs are gone, giving actual lows in the middle of the night. Would like to automate TT for this as well, and/or profile change for first 1-2 hours after site change to give more insulin up front instead of SMB's after the rise already happens. (Example: slow insulin absorbtion from evening meal at 10PM/22.00, several SMB's, sharp drop followed by overcorrection, slight high, more SMB's when automated hypo tt runs out, then new low in the morning)
[image]

sabsoubi, 2024-02-01:
I support this small enhancement. A new trigger "new CGM sensor insert" would be very useful.
Please consider this enhancement for the next release. Many thanks

MilosKozak, 2026-10-11:
AAPS automation now has "Sensor age" and "Cannula age" triggers. You can make a rule like "sensor age less than 1 h" that starts a temp target or a profile switch.

closed 2026-10-11 by MilosKozak