nightscout/AndroidAPS #2012: Master 3103 / Treatments menu / careportal / 3 dot menu not clear

robertrub, 2022-08-14:
[image]

Not very clear what this option does...

Andries-Smit, 2022-08-14:
It remove start events from the log.
Is your suggestion to rename it? (Sugestions are welecome)
 Or should it be better documented?

robertrub, 2022-08-14:
Ok, tnx. You don't have too much space for a better text :(

_Delete "AAPS started" entries_

Would be great if it is not too long. If it is too long, without the _entries_ word but keep the quotes.

Andries-Smit, 2022-08-14:
Good suggestion, it is more descriptive

MilosKozak, 2022-08-15:
https://github.com/nightscout/AndroidAPS/commit/551c927495224a7e0622252cc3282696d4613a37

closed 2022-08-15 by MilosKozak