nightscout/AndroidAPS #2014: [Boost 3.1.0.2] [Boost 3.1.0.2] SMB's are calculated but only very sporadically switched through to the pump

Thomas-Wolff, 2022-08-14:
after the update to Boost 3.1.0.2, calculated SMB's are only very sporadically switched through to the pump. Boost TAB SMB's always activated and are verified here also clean and pass to the loop tab. In the Loop TAB basal changes are cleanly put through to the pump, but the SMB's stand to 99% always with the message 'wait for pump result'.
See screenshots and LOG files.

[image]

[image]

[AndroidAPS._2022-08-14_08-12-04_.14[1].zip](https://github.com/nightscout/AndroidAPS/files/9333025/AndroidAPS._2022-08-14_08-12-04_.14.1.zip)
[AndroidAPS._2022-08-14_08-12-04_.15[1].zip](https://github.com/nightscout/AndroidAPS/files/9333027/AndroidAPS._2022-08-14_08-12-04_.15.1.zip)

With best thanks Thomas

MilosKozak, 2022-08-14:
here is supported on latest master and dev. ask Boost devs

closed 2022-08-14 by MilosKozak

Thomas-Wolff, 2022-08-14:
Hello Milos, thank you very much.

I was in contact with Tim2000s yesterday, but unfortunately he didn't know what to do. We will contact him again in a moment.

Greetings