nightscout/AndroidAPS #2017: Data is coming from different pump

camtech57, 2022-08-15:
Combo Pump
AAPS Version 3.1.0.3
Phone KK2 mini
Happens every day 
Switched to virtual pump and back again
Restarted phone
Reinstalled AAPS
Did not solve issue

[image]

justmara, 2022-08-16:
I confirm, this alert appears on every app start. Pretty annoying.

RyszardS, 2022-08-17:
Hi I have the same issue with combo pump after update to 3.1.0.3

RyszardS, 2022-08-17:
Hi, I found in discard AAPS channel that it could be related with Google play protect, so I disable this option and I switch to virtual pump( select dana R) then I switch back to compo pump. After that loop start working again

justmara, 2022-08-17:
@RyszardS this always starts working when you simply dismiss alert. And then in comes back after app restart.

RyszardS, 2022-08-17:
I will leave feedback after couple of days

cascer1, 2022-08-18:
I don't think disabling Google Play Protect is a valid solution to this problem.

databoy2k, 2022-10-02:
Issue applies to Virtual Pump type as well. I have to switch to a real pump and then back to Virtual.

AAPS 3.1.0.3 on a Pixel 6, Android 13

MilosKozak, 2022-10-03:
hopefuly resolved by https://github.com/nightscout/AndroidAPS/commit/3f3cdde53bcff139515cf9cdfb9bb5dbd68a4ca2

ThomasWiesner, 2022-10-12:
Hi there, sorry to reopen this issue, but we have the same error with combo, but AAPS version 3.0.1. 
Happens always when combo is chosen, but not with virtual pump. No loop possible at the moment. What can we do?
Thank you for any help, my wife is more than desperate...
Thomas

justmara, 2022-10-12:
And I can confirm that this aint fixed. Running dev build with Combo pump and see that alert after (literally) every aaps restart.

MilosKozak, 2022-10-13:
i need logs from latest dev

justmara, 2022-10-15:
[AndroidAPS (22).log](https://github.com/nightscout/AndroidAPS/files/9792303/AndroidAPS.22.log)
here is log since app start
sorry, im running russian locale on my kid's phone so the notification has text 'Данные поступают'. its first appear is at `17:15:48.474`

justmara, 2022-10-17:
Ouch, new driver? Need testers? I have spare old Combo to be used with second phone ;)

databoy2k, 2022-10-21:
Let me know if this should be broken into its own issue, but the virtual pump seems to want Ruffy as well. On AAPS start I get the following:
`12:16:53.238 [main] I/PUMP: [RuffyScripter.<init>():178]: No connection to ruffy. Pump control unavailable.`

After starting AAPS, but if I don't change my pump from virtual to anything else, my bolus log looks like this:
```
12:28:19.523 [Thread-115] D/PUMP: [VirtualPumpPlugin.deliverTreatment():186]: Delivering treatment insulin: 4.0U carbs: 0.0g Success: true
Enacted: true
Comment: OK
Insulin: 4.0 U
12:28:19.526 [RxCachedThreadScheduler-20] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: Data is coming from different pump. Change pump driver to reset pump state.
12:28:19.527 [Thread-115] E/PUMP: [PumpSyncImplementation.confirmActivePump():83]: Ignoring pump history record  Allowed: 10/21/22 05:25:28AM MDI eVDUFdj6SRm4-YB6QBtehX Received: 1666376890283 10/21/22 12:28:10PM MDI 
12:28:19.527 [Thread-115] D/PUMPQUEUE: [CommandQueueImplementation.bolus$lambda-10():241]: Going to store carbs
12:28:19.529 [Thread-115] D/PUMPQUEUE: [CommandBolus.execute():30]: Result success: true enacted: true
12:28:19.529 [main] D/PUMP: [BolusProgressDialog.onResume$lambda-3():130]: Running id 1666376890283. Close request id  1666376890283
12:28:19.529 [RxCachedThreadScheduler-50] D/WEAR: [DataLayerListenerServiceMobile.sendMessage():173]: sendMessage: /rx_bridge {"type":"info.nightscout.shared.weardata.EventData.BolusProgress","percent":100,"status":"Success"}
12:28:19.534 [RxCachedThreadScheduler-76] D/DATABASE: [CompatDBHelper.dbChangeDisposable$lambda-18():44]: Firing EventTreatmentChange 1666377490283
```

But once I've changed my pump from virtual to dash and then back again, it looks like this:
```
12:31:26.213 [Thread-147] D/PUMP: [VirtualPumpPlugin.deliverTreatment():186]: Delivering treatment insulin: 4.0U carbs: 0.0g Success: true
Enacted: true
Comment: OK
Insulin: 4.0 U
12:31:26.214 [Thread-147] D/PUMP: [PumpSyncImplementation.confirmActivePump():69]: Registering new pump MDI eVDUFdj6SRm4-YB6QBtehX
12:31:26.227 [RxCachedThreadScheduler-66] D/DATABASE: [CompatDBHelper.dbChangeDisposable$lambda-18():49]: Firing EventTreatmentChange 1666377075123
```

Seems like an issue where it only registers a new pump serial number if the pump has been reset. Maybe AAPS should register a new pump serial number on start?

MilosKozak, 2022-10-22:
> Let me know if this should be broken into its own issue, but the virtual pump seems to want Ruffy as well. On AAPS start I get the following: `12:16:53.238 [main] I/PUMP: [...]

In dev SN is ignored for VirtualPump. It should not happen anymore

justmara, 2022-11-04:
This is really a big problem. This bug makes AAPS ignore boluses from pump history
```
13:36:30.286 [Thread-33] D/PUMP: [RuffyScripter.runCommand():263]: Attempting to run cmd: BolusCommand{bolus=3.5}
13:36:30.287 [Thread-33] D/PUMP: [RuffyScripter.runCommand():278]: Connection ready to execute cmd BolusCommand{bolus=3.5}
...
13:36:45.015 [BolusCommand] D/PUMP: [BolusCommand.execute():170]: Delivering bolus, remaining: 3.5
...
13:37:02.176 [BolusCommand] D/PUMP: [BolusCommand.execute():192]: Stage 4: full bolus of 3.5 U was successfully delivered
...
13:37:03.959 [Thread-33] E/PUMP: [PumpSyncImplementation.confirmActivePump():83]: Ignoring pump history record  Allowed: 04.11.2022 13:36:05 Accu-Chek Combo feDCRjrNSy6Dq9Av5QAZ5t Received: 1667558100000 04.11.2022 13:35:00 Accu-Chek Combo feDCRjrNSy6Dq9Av5QAZ5t
```
This shows the case when Pump delivered large bolus of 3.5, but it were ignored by AAPS making it count low IOB and keeping delivering more SMB afterwards. Thats a really dangerous problem.
See more in log attached

[AndroidAPS (25).log](https://github.com/nightscout/AndroidAPS/files/9938252/AndroidAPS.25.log)

MilosKozak, 2022-11-04:
@justmara version tested?

justmara, 2022-11-05:
@MilosKozak that log is from 3.1.0.3-master build. must people uses _stable_ version, and this issue is pretty dangerous. dont you think its a reason for 3.1.0.4 bugfix release?

olorinmaia, 2022-12-06:
Following. Testing virtual pump and this error happens daily. Changing pump type back and forth fixes it. Im using 3.1.0.3. Should i upgrade to the bugfix version or wont this happen when i activate a real omnipod dash?

CaioBauab, 2022-12-19:
Same error at every bolus using MDI. Even changing from MDI to Virtual APS pump and changing back doesn't solve anymore the bug. I need to register using insulin button (not deliver, register only). Any news about a fix?

marutasan, 2023-01-10:
@MilosKozak I'm getting this same error with an Accu-chek insight pump. I've had to delete the pairing with my previous Insight pump and pair a new one (the old one stopped working), and now this is happening. As @justmara said, everytime I deliver a bolus this error appears and the bolus is not being written to AAPS db, dangerous. As my case is a little bit different (change of pump, everything else the same), is there any other solution? Thanks!!!!

cascer1, 2023-01-11:
This issue is resolved for me when using the dev branch and new combo driver.

marutasan, 2023-01-11:
I cannot enter the facebook group yet, I'm waiting to get admitted. Do any of you know if my case can be solved easily? It all worked perfectly well until I had to unpair a pump and pair a new one. The message "Data is coming from a different pump" in fact it's true, but it should admit this new pump's data if I've unpaired the old one, shouldn't it? Two hypos this last night, because the algorithm doesn't have the info re what's the pump doing (boluses or TBRs...).

I couldn't find any info on the docs about this either...

Please, help...

marutasan, 2023-01-11:
@MilosKozak I've tried building latest version to see if that solves the problem, but I can't. I get a StackOverFlow error. Nothing worked: clean project + build again, unistall completely Android Studio and installing again from scratch and building again from scratch...  nothing

> Task :app:compileFullReleaseJavaWithJavac
compiler message file broken: key=compiler.misc.msg.bug arguments=11.0.13, {1}, {2}, {3}, {4}, {5}, {6}, {7}
java.lang.StackOverflowError
	at jdk.compiler/com.sun.tools.javac.code.Types$15.visitClassType(Types.java:2465)
	at jdk.compiler/com.sun.tools.javac.code.Type$ClassType.accept(Type.java:993)
	at jdk.compiler/com.sun.tools.javac.code.Types$UnaryVisitor.visit(Types.java:4937)
	at jdk.compiler/com.sun.tools.javac.code.Types.supertype(Types.java:2462)
	at jdk.compiler/com.sun.tools.javac.comp.Resolve.findInheritedMemberType(Resolve.java:2194)
	at jdk.compiler/com.sun.tools.javac.comp.Resolve.findMemberType(Resolve.java:2231)
	at jdk.compiler/com.sun.tools.javac.comp.Resolve.findInheritedMemberType(Resolve.java:2196)
	at jdk.compiler/com.sun.tools.javac.comp.Resolve.findMemberType(Resolve.java:2231)
	at jdk.compiler/com.sun.tools.javac.comp.Resolve.findInheritedMemberType(Resolve.java:2202)
	at jdk.compiler/com.sun.tools.javac.comp.Resolve.findType(Resolve.java:2296)
	at jdk.compiler/com.sun.tools.javac.comp.Resolve.findIdentInternal(Resolve.java:2355)
	at jdk.compiler/com.sun.tools.javac.comp.Resolve.findIdent(Resolve.java:2341)
	at jdk.compiler/com.sun.tools.javac.comp.Resolve.resolveIdent(Resolve.java:2602)
	at jdk.compiler/com.sun.tools.javac.comp.Attr.visitIdent(Attr.java:3490)
	at jdk.compiler/com.sun.tools.javac.tree.JCTree$JCIdent.accept(JCTree.java:2248)
	at jdk.compiler/com.sun.tools.javac.comp.Attr.attribTree(Attr.java:655)
	at jdk.compiler/com.sun.tools.javac.comp.Attr.attribType(Attr.java:715)
	at jdk.compiler/com.sun.tools.javac.comp.Attr.attribType(Attr.java:708)
	at jdk.compiler/com.sun.tools.javac.comp.Attr.visitTypeCast(Attr.java:3418)
	at jdk.compiler/com.sun.tools.javac.tree.JCTree$JCTypeCast.accept(JCTree.java:2022)
	at jdk.compiler/com.sun.tools.javac.comp.ArgumentAttr.visitTree(ArgumentAttr.java:207)

marutasan, 2023-01-11:
@MilosKozak , I've been able to upgrade to version 3.1.0.3 from version 3.02 and still not working. There must be some way to tell AndroidAPS that the new pump is the good one, and that it should accept it's data, but I don't know how. Still getting "Data is coming from different pump" error all the time :(

MilosKozak, 2023-01-11:
does this appear in latest dev?

marutasan, 2023-01-11:
Hello @MilosKozak , I've always used the main branch, I understand you want me to try with the dev branch because there are some changes related to this issue? So the Insight driver has also changed, not only the Combo one?

marutasan, 2023-01-11:
@MilosKozak I finally could get into the facebook group. I saw people suggesting to switch to any other pump and then back to your pump, and it worked. I don't know how many hours I've spent with this today :-(
It'd be great if this was in the documentation on the website, just a suggestion!
Thanks a lot,

@marutasan

Philoul, 2023-01-11:
@marutasan and @MilosKozak 

I just tested again this evening with latest dev (3.1.0.3-dev-f) and 2 different Insight pumps on my dev config (first one with firmware v3 and second one with firmware v2...), but I could not reproduce this issue...

Test done:
1) Pair Pump 1
2) Deliver 2 manual boluses (1.00U then 0.50U) recorded correctly in database
3) UnPair Pump 1 then Pair Pump 2
4) Deliver a manual bolus (2.00U) also recorded correctly in database without any alert
5) UnPair Pump 2 then Pair again Pump 1
6) Deliver a third different manual Bolus (3.00 U), no alerts, everything worked fine
[image]

On my side everything worked fine without any alerts... 
It's very strange because I remember I got this issue in the past during the update of Insight driver for Room database, but it was fixed several months before the v3 release, and I never got again this issue...

Philoul, 2023-01-11:
After my previous test, I did the same with the Insight Driver in kotlin... And I couldn't reproduce the issue.
=> All boluses delivered and recorded correctly in database, without alerts with Pump 1, then Pump 2 and then again with Pump 1...

jotomo, 2023-06-10:
@justmara 
1) What's your ruffy version?
2) Do you still get spurious _new pump connected_ messages?
3) Could you test a PR with a possible fix?

justmara, 2023-06-10:
@jotomo sorry, but we've switched to omnipods ages ago, so now can't test it

MilosKozak, 2026-10-11:
This came from the old Ruffy-based Combo driver, which reported a made-up serial number. That driver was removed in 2023. The current driver (combov2) uses the pump ID from the pairing data, which does not change after a restart. Closing as obsolete. If you see this message with a current version, please open a new issue with the log and the pump you use.

closed 2026-10-11 by MilosKozak