nightscout/AndroidAPS #2019: Play Protect scan kills connection to Combo

tKostka, 2022-08-16:
Every scan of google play protect (manual or automatic triggered) kills the connection from AAPS to Combo on my new phone. Only way to re-establish connection to Combo is a forced close of AAPS and restart of AAPS, or restart of phone. This issue is reproduceable, happens every time: trigger play protect scan, give bolus in AAPS -> error message in AAPS

AndroidAPS 3.1.0.3
Nothing Phone (1) with Android 12
I did not have this issue on my old phone - Oneplus 7t with Android 11. There, AAPS and xDrip are working with play protect on.

I now, the workaround is to switch off play protect, but when using an additional work profile on android (to connect to company mails and services), play protect can't be switched off (switch greyed out). I can live without that, but would be easier to have everything on one phone.
This is definitely not a problem for most of the users, but is there a possibility for improvement? Would be great.   
 
Attached a logfile. play protect scan triggered 20:22, Bolus given 20:23

```
20:23:14.750 [Thread-82262] E/PUMP: [RuffyScripter.runCommand():360]: CommandException while executing command
info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.commands.CommandException: Unexpected exception while initiating/restoring pump connection
	at info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter.ensureConnected(RuffyScripter.java:512) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter.runCommand(RuffyScripter.java:277) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter.readPumpState(RuffyScripter.java:238) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ComboPlugin.runOnConnectChecks(ComboPlugin.java:932) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ComboPlugin.runCommand(ComboPlugin.java:874) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ComboPlugin.deliverTreatment(ComboPlugin.java:512) ~[na:0.0]
	at info.nightscout.androidaps.queue.commands.CommandBolus.execute(CommandBolus.kt:26) ~[na:0.0]
	at info.nightscout.androidaps.queue.QueueThread.run(QueueThread.kt:131) ~[na:0.0]
Caused by: android.os.DeadObjectException: null
	at android.os.BinderProxy.transactNative(Native Method) ~[na:0.0]
	at android.os.BinderProxy.transact(BinderProxy.java:571) ~[na:0.0]
	at org.monkey.d.ruffy.ruffy.driver.IRuffyService$Stub$Proxy.doRTConnect(IRuffyService.java:191) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter.ensureConnected(RuffyScripter.java:488) ~[na:0.0]
	... 7 common frames omitted
```

MilosKozak, 2022-08-17:
yes it's mentioned in wiki
https://androidaps.readthedocs.io/en/latest/search.html?q=play+protect&check_keywords=yes&area=default

MilosKozak, 2022-08-17:
the problem is comming from self signed certificate which is evaluated as suspicious by play protect. there is nothing we can do with it

closed 2022-08-17 by MilosKozak