nightscout/AndroidAPS #2030: Using Nsclient results in duplicate profileswitch events in NS and in AAPS

wootmasterslick, 2022-08-19:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused the problem

18 aug 2022 23:22

Upon enabling nsclient the already present profileswitch events become duplicated and are the uploaded to NS.
Sometimes numerous copies form and impact functioning of the app.
Only the profileswitches are affected. They are flooding the NS and AAPS with useless information and making the app less responsive. The times are sometimes also varying so that affects any autotune calculations one might do.
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).

3.1.0.3

Pump accu Check Insight

- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html

log files are coming asap

- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

done

Screenshots:

[image]

[image]

[image]

[image]

Discussion:
https://discord.com/channels/629952586895851530/629952669967974410/1009953469127458896

Philoul, 2022-08-19:
No impact on Autotune calculation done with AAPS plugin: for each BG value, Autotune use the active profile got from database to calculate basal rate (and associated IOB, Activity...)

wootmasterslick, 2022-08-19:
> No impact on Autotune calculation done with AAPS plugin: for each BG value, Autotune use the active profile got from database to calculate basal rate (and associated IOB, Activity...)

Do you not agree it is an issue?

Philoul, 2022-08-19:
I think having 3 or more lines for each profile switch is an issue, we should have only 2 lines (one line for the request, and a second line for the profile switch done into the pump).
But even if there is a bug, there is no impact on Autotune calculation (cf comment below in Issue's description)
> The times are sometimes also varying so that affects any autotune calculations one might do.

MilosKozak, 2022-10-04:
this should be resolved in AAPS dev and NS 14.2.6

closed 2022-10-04 by MilosKozak