nightscout/AndroidAPS #2032: [3.1.0.3] Future insulin activity projection line disappears

nlof, 2022-08-20:
It happens recurrently, I just saw it again at 13:58 today. The yellow insulin projection line into the future disappears sometimes. Didn't use to happen with v. 2.8.2.1. It's a dashed line and most of the time it appears normally on 3.1.0.3 as well.

It may have to do with existing text from treatments around that area? (but not sure).
Logs and 2 screenshots attached.

[AndroidAPS._2022-08-20_12-52-10_.5.zip](https://github.com/nightscout/AndroidAPS/files/9386929/AndroidAPS._2022-08-20_12-52-10_.5.zip)
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/9386930/AndroidAPS.log)
[image]
[image]

nlof, 2022-08-20:
And it partially reappeared a few minutes later.

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/9386956/AndroidAPS.log)

[image]
![Screenshot_2022-08-20-14-13-50-00 (https://user-images.githubusercontent.com/70835405/185743455-8efdc564-6f02-4cb4-89b3-02922d08b6cf.jpg)

Philoul, 2022-08-21:
I think you only have only forecast for activity until. the next hour in future.
If you are just before a new hour (13h58) or graph is still not fully updated (14h04), you don't see the forecast activity, and a bit later, you see it up to next hour (15h in your screenshots)...

nlof, 2022-08-21:
You're absolutely correct, I followed this for a while and reached the same conclusion. 
It's still a bug, there's no reason for the graph to stop at the next round hour: it works fine on v 2.8.2.1, I'll post a screenshot.

MilosKozak, 2022-12-04:
https://github.com/nightscout/AndroidAPS/commit/1f1a2eae48564779a9f88363749f6d58fa8441f1

closed 2022-12-04 by MilosKozak