nightscout/AndroidAPS #2034: MINOR / Watchface not centered

robertrub, 2022-08-21:
@Andries-Smit On my Fossil gen 5 watch, this AAPS face is not centered and with larger numbers (20 and over) for the hour, one part of the number is cut off. The same face from xDrip is centered.

[image]

robertrub, 2022-08-21:
Feature request : Same watchface in xDrip has the date also ;)
The date is often missing from AAPS faces (I didn't test them all ...)

Andries-Smit, 2022-08-21:
Most watch faces where designed for square watches. Some of the round watches are an afterthought.
I'll review them to see if I can fix it. Though making it scale across all shapes and size a lot of work

rICTx-T1D, 2022-08-21:
Fossil smartwatch's re-positon the watchface each minute and protect the watch for burn-in effects.  When you going in active mode, all is centered correctly on my Fossil  carlyle (Gen5).

robertrub, 2022-08-21:
That's not the case on my Fossil. Other AAPS and xDrip faces are centered and don't move either.

PR #2307 (2022-12-29): Wear: unify watch face round and square layout
- Make watch faces use Constraint layouts, so they are responsive regardless of the screen size. Improves centring content. Fixes #2034
- Use AM/PM indicator `timePeriod` to make the 12h time notation fit on the screen.
- Use dimension file based on screen size qualifier to size the watch face font sizes. 
- Remove duplicate layout. For easier maintenance
- Let digital style background be selected by resource qualifier
closed 2023-08-22 by robertrub