nightscout/AndroidAPS #2036: [3.1.0.3] Sluggish performance / Large amounts userdata

I1985, 2022-08-21:
I've been very happily using AAPS very happily for about 3 years now and can't even imagine how I managed before I started looping.
The last 2 years have been on my current phone a Google Pixel 4a and that always worked well. However after updating to 3.1.0.3 on August 6th (from 3.0.0.1) I feel like the battery drains a bit faster, but I've not been at work like normal so it was hard to compare.

But now since about a week I feel like AAPS is getting sluggish. It really takes some time to process BG Deltas when a new value comes in for instance. After a tip on FB I checked the app info in the phone settings menu, where it shows that AAPS uses 1gb of user data.
That seems like a lot to me? Can't really imagine what would take that much space?

I do have plenty of storage remaining (over 50gb unused). But I can imagine it taking some time having to manage all that data. En could potentially also cause the slow response to new data and perhaps the extra battery drain?
Came across some complaints about phones overheating after updating to 3.1.0.3 and wonder if these cases might be related? Although I haven't noticed this myself.

As far as I remember I did a clean 2.7.0 install when I got the phone 2 years ago. After that always updated (although I missed a few smaller updates).

Could really find this issue seasoning, but if it was reported already my apologies.

If logs would be appreciated just let me know which you need.

I1985, 2022-09-05:
I made a fresh backup of my settings and then I bit the bullet and cleared all the storage for AAPS in the android app settings. Rebooted the phone, imported the settings, set permissions reconnected pump etc.

AAPS is working well now, seems much quicker. The update bar on the bottom would take at least 5-10 seconds before, now its more just a quick flash instead of a progress bar.

Before this cleanout the user data was 1gb
After this cleanout the user data is 2mb 

And everything seems to work normally (just much quicker)

Downside of this is that info like IOB, COB and the timers are lost. So if others want to do this, time it well.

I've seen/heard others about similar issues. Seems worth looking into?

MilosKozak, 2022-12-04:
automatic database cleanup is in place now

closed 2022-12-04 by MilosKozak