nightscout/AndroidAPS #2038: Processing a big Nightscout client queue limits loop function

Andries-Smit, 2022-08-23:
My NS client was paused for a month. This caused a big queue which took a day or more to process; during that time, the battery was draining. In my experience, the loop functioned not as expected. And the app did not respond well.

A similar issue was reported on Facebook:
> While it is uploading, my closed loop seems to be paused. Not talking about the chart updating the current values, but indeed I'm ? just running with my preset basal and not having any reaction from the system: neither tbr nor smb - so BG rose high.

- Should there be a Nightscout client API for batch upload?
- Should an option in the menu to clear the queue?

Philoul, 2022-08-26:
I conform there are some issues during NS synchro:
I limited during about 1 days Sync settings to save battery, and yesterday morning I enabled again to synchroniźe NS (about 1300 item in queue).
- During synchro loop was not working (more than 1 hour duration for the whole synchro).
- I got a fake Profile switch in graph (the profile switch done by autotune at 4 AM, was synchronized in NS later, and the white star is also visible in graph for the NS sync5 hours later.
- the number of items in queue during sync was strange (it starts to decrease, I saw about 700 item about 10min after sync start, then it increase up to more than 1600 items in queue, and finally drop down to 0).

[image]

robertrub, 2022-10-26:
In dev-c, I "cleaned" (deleted older than 3 months data) in AAPS. It creates (a bug) the re-syncs all the data (cue of 55000)...

After 3 something hours, all the data was sent to NS. The loop was working correctly while the upload was going on.

Maybe your bug was solved or maybe a fast phone/internet/??? lets enough CPU time to calculate the loop.