nightscout/AndroidAPS #2039: [3.1.0.3] Glucose projection lines do not get redrawn immediately sometimes

nlof, 2022-08-23:
Around 3:05-3:06 PM today.
After giving manual insulin bolus (in virtual pump (eros) but I think also in regular omnipod eros IIRC), the glucose projection lines do net get redrawn immediately even if SMB and the Loop tabs get refreshed manually. 
This occured multiple times on trials. They do get redrawn only if I add carbs (even 1 gram). And they do get redrawn eventually on next BG update.
I didn't encounter this issue in v. 2.8.2.1 IIRC.

Atraching logs and screenshots.
[AndroidAPS._2022-08-23_15-03-48_.1.zip](https://github.com/nightscout/AndroidAPS/files/9402420/AndroidAPS._2022-08-23_15-03-48_.1.zip)
[AndroidAPS._2022-08-23_15-03-48_.0.zip](https://github.com/nightscout/AndroidAPS/files/9402421/AndroidAPS._2022-08-23_15-03-48_.0.zip)
[image]

nlof, 2022-10-09:
It may take a while as I need to use a friend's computer to build the app...

MilosKozak, 2023-01-21:
improved in dev. data is available after OAPS run

closed 2023-01-21 by MilosKozak