nightscout/AndroidAPS #2041: [FEATURE REQUEST] Read data from smartpen Novopen Echo Plus/Echo Pen 6

rICTx-T1D, 2022-08-23:
New smartpens with nfc interface was released in Europe.  So diasend and glooko can read out the last injection with timestamp and amount. Please integrate it in AAPS, so MDI user or other can use it to write notices.

Andries-Smit, 2022-08-24:
The Echo pen is supported by xDrip. Can you get the data via xDrip/nightscout?

rICTx-T1D, 2022-08-25:
At my last knowledge,  only some "official" apps in playstore support the nfc interface from the new smartpens.

Andries-Smit, 2022-08-25:
Looking in the config, there might already be something experimental, did you try it?
https://github.com/NightscoutFoundation/xDrip/blob/master/app/src/main/res/xml/xdrip_plus_prefs.xml#L1081

Andries-Smit, 2022-08-25:
The related topic of your link, they are testing... 
https://github.com/NightscoutFoundation/xDrip/issues/2074

rICTx-T1D, 2022-08-25:
Hi @Andries-Smit ,
at moment i make MDI with virtual pump in AAPS and the BYODA . I dont use xdrip on my phone.

Is it possible to merge/clone the code to AAPS? Maybe a pump user will write a extra dose of pen insulin in ns-record.

Andries-Smit, 2022-08-26:
You might be able to use xDrip just for the pen interface. As MDI is not the primary goal of AAPS I
would not opt for duplicating the feature, if xDrip can do it. Though there might be someone to implement the feature.

jamorham, 2022-08-28:
This feature is known to be working well within xDrip and I would expect that its possible to use the pen scanning without using any live glucose collector within xDrip although I haven't tested this. Data will be uploaded to Nightscout and I can arrange a local broadcast if that would be helpful as well.

MilosKozak, 2022-09-27:
this is not purpose of AAPS, sorry
lot of code, marginal impact

closed 2022-09-27 by MilosKozak