nightscout/AndroidAPS #2044: Autosens / Weighted Average - Sensitivity stuck / not resetting to 100% when there are COB

janvitos, 2022-08-24:
Hi,

Today, I noticed something odd about Autosens / Sensitivity Weighted Average and I think it might be due to a bug.

This afternoon, my son’s insulin sensitivity went up from 100% to 120% (see screenshot below) and Autosens adjusted its multiplier to 1.2 accordingly, as seen under “Script debug.” I could also see other altered values like ISF, which had been lowered to 120 from 144.

A bit later, my son started playing outside and became much more sensitive to insulin, but Autosens remained at 120%. Again, I noticed the Autosens multiplier was still at 1.2 and other values were still altered.

We were having a bit of a hard time getting him out of hypo, and I suspect Autosens might’ve been partly to blame since it was stuck at 120% even while his sensitivity was clearly much lower.

I then decided to test Sensitivity Oref1 to see if the behaviour was the same or different. As I thought, Sensitivity Oref1 reverts back to 100% when there are COB, which confirms there's most likely a bug in the Sensitivity Weighted Average plugin.

[image]

justmara, 2022-08-25:
Read the docs: 

> If continuously eating carbs over an extended period, autosens will be less effective during that period as carbs are excluded from BG delta calculations.

janvitos, 2022-08-25:
> Read the docs

I did read, and re-read the docs.

I truly believe there is a bug here.

justmara, 2022-08-25:
It always were acting like this - sticking with latest calculated value until new one CAN be calculated. While COB>0 it has no new data, so new calculations cannot be done -> no sensitivity update.

janvitos, 2022-08-25:
> It always were acting like this - sticking with latest calculated value until new one CAN be calculated

So why is Oref1 resetting to 100% when COB kick in? Shouldn't Weighted Average and Oref1 have the same behaviour regarding COB?

nlof, 2022-08-28:
Weird, I didn't pay attention to such an issue.
I just checked in the recent graph and if you look at the red marked area, despite having COB, the white autosens (sensitivity) line is decreasing... 
I've been running the weighted average option with a timeframe of 4 hous for a year now....
Yes, sometimes it's somewhat slow to catch up with the decrease, but it's the nature of the beast...

[image]

janvitos, 2022-08-28:
I'm going to close this as different people seem to have different results.

closed 2022-08-28 by janvitos

nlof, 2022-08-29:
Unsure whether you should rush and close it, I just don't remember seeing this for sure. It's worth continuing to monitor it as it is possible that sensitivity doesn't decrease in real time but maybe it's recalculated later? I don't know for sure, I'll try to watch it in real time if I remember 🙂

nlof, 2022-08-29:
Yesterday again sensitivity changed with COB (both down and up, see screenshot), but again, not sure i remember if it happened in real time. Probably yes, but unsure.
[image]