nightscout/AndroidAPS #2048: AAPS v3.1.0.2 no longer syncs pump manual bolus history for IOB calculations

muthijammy, 2022-08-25:
- Issue started only after v3.1.0.2
- has happened twice already
- mobile switched off. Latest issue at 3:20 pm bolus 1 unit directly on combo pump as shown below
[image]

- Mobile switched on at 6 pm. 1u not used in IOB calculations or treatment tab
[image]
[image]

- Please note 1u does showup in combo tab last bolus field but not used elsewhere
[image]

 - Not sure if limited to combo pump or all pumps 

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/9422863/AndroidAPS.log)

dv1, 2023-11-16:
This should be taken care of in the new combov2 driver, since the command-mode history delta will contain the events that happened since the last time the delta was retrieved. In this case, it should contain an entry that shows that a bolus was manually requested. The history delta has a limited space for entries, though normally, this limit should not be reached (you'd have to deliver lots of manual boluses in a short amount of time for example).

MilosKozak, 2026-10-11:
The old Ruffy-based Combo driver was removed in 2023 (first in 3.3.0.0). The current driver (combov2) reads the pump history on every connect and syncs manual and quick boluses to AAPS, so they count for IOB. Closing as obsolete. If a manual bolus is still missing with a current version, please open a new issue with the log.

closed 2026-10-11 by MilosKozak