nightscout/AndroidAPS #2050: Bolus requested through WearOS are not executed

niwreg-coder, 2022-08-26:
After confirming the desired bolus on WearOS, I get a toast indicating 'treatment requested', but the treatment is never executed while I am close to my Phone. FYI, I am using the Omnipod Dash and am on Galaxy Watch 4 and the S22 and run the most recent operating systems and the most recent versions of AndroidAPS for both my phnoe and watch.

I was trying to explore this issue, but I am to illiterate in Kotlin to make sense of this. Yet, my idea is that it is going wrong at line 76 where EventWearToMobile is called: [](https://github.com/nightscout/AndroidAPS/blob/f8acbad759ccdc4dcdc58a7d150103dfe757ff94/wear/src/main/java/info/nightscout/androidaps/interaction/actions/TreatmentActivity.kt)

Andries-Smit, 2022-08-26:
@niwreg-coder just to share the happy flow after the "treatment Requested" toast, the bolus request is send to the phone, when allowed a confirmation message is send. The confirmation is shown ontgelden watch, swiping right to left to and click V to confirm.
Do your receive the confirmation message?

niwreg-coder, 2022-08-26:
@Andries-Smit Thank you for clarrifying on such a short notice. Really greatful for all the people dedicating their time to this project. What does 'allowed' mean in this context? Does this mean 'allowed within certain predefined boundaries' such as maximum bolus size, or does allowance involve an additional manual step?

I do not receive a confirmation message on the watch nor see anything happening in the app on my phone.

JimZam, 2022-08-26:
@niwreg-coder If you don't have an AAPS watchface it will not work, I have spent hours reseting and reinstalling until someone at Discord confirmed that behaviour...

niwreg-coder, 2022-08-27:
It is working now. The only thing that I can think of that has changed is that I passed objective 3. Are these objectives in any way linked to the ability of the user to administer boluses through WearOS?

MilosKozak, 2022-09-01:
it's indepedent thing

closed 2022-09-01 by MilosKozak