nightscout/AndroidAPS #2052: Watchface, widget and notifications take time to update

JimZam, 2022-08-26:
Good evening. Whenever AAPS receives xDrip's data it updates, but widget, watchface and notification (the one you see when you wipe from the top of the screen) takes some time (sometimes 1-3 minutes, and the watchface even 15-20 minutes in the worst of the cases, what makes it useless). That makes me receive an alarm from xDrip and see a different value in the watch. To avoid it I changed the watchface to xDrip, that updates inmediatly, but then I can't use treatments from the watch. I can't say if that happened in previous versions as I changed recently my watch, but in 2.8.2.1 it worked fine.
Both phone and watch are in version 3.1.0.3.
Thank you for your help.

JimZam, 2022-09-03:
That's what I said, any help?
Thank you
[image]

I1985, 2022-09-05:
Hey Jim,
That sounds sort of similar to what I was experiencing. Might be worth checking how much user data the app has stored?
See [#2036](https://github.com/nightscout/AndroidAPS/issues/2036#issue-1345584057)

JimZam, 2022-09-05:
Hello. It seems it has worked, but now I don't have predictions in my watchface although thery appear in the phone, maybe I have to wait longer? :-(

I1985, 2022-09-06:
Glad that solved the issue Jim. Thought it sounded a lot like what I had.

closed 2022-10-04 by MilosKozak

JimZam, 2022-10-11:
Hello. Thank you for your help, I'll give it try.
Regards