nightscout/AndroidAPS #2057: Stacking of TT definitions does not work in Version 3.x

ga-zelle, 2022-09-01:
Let's assume I have an active TT runnig with 90 minutes left and I superimpose a new one for 10 minutes. Once the 10 minutes are over AAPS2.x used to return to the previous TT for its remaining 80 minutes. However, with 3.0 and 3.1 it just cancels the underlying TT also.

Thus stacking of time windows still works for profile percentage changes.

Andries-Smit, 2022-09-02:
Is this duplication of https://github.com/nightscout/AndroidAPS/issues/1282 ?

ga-zelle, 2022-09-03:
Yes, I did not see that when I scanned through.
Still, I think it worked that way before and is desirable. For example I could could enable  SMB by TT=90 for 30 mins and then return to TT=101 to stop SMB again.

Internally, routines to handle TTs of different activation times seem availble.

swissalpine, 2022-09-05:
I also miss this previous functionality!

nlof, 2022-09-11:
This is the exact opposite of the situation now with profile changes!
 
In the past (with v. 2.8.2.1), if a long temp profile change was overridden by a shorter temp profile change of even 10 minutes, the profile would reset to 100% after the 10 minutes in this case.

With v. 3.1.0.3, after the 10 minutes temp profile change, the profile would go back to the earlier temp profile instead of resetting to 100% for the remaining time of the longer, earlier temp profile.

See https://github.com/nightscout/AndroidAPS/issues/2053

closed 2026-10-15 by MilosKozak