nightscout/AndroidAPS #2058: Missing alignment of messages inside Script Debug section of the SMB-tab

ga-zelle, 2022-09-01:
All messages from determine-basal.js are patched together end-to-end without any line breaks. This started in 3.0 dev versions, was (nearly) fixed in 3.0.0.1 and now happens again in 3.1.0.2.

Andries-Smit, 2022-09-02:
@ga-zelle could you please add screenshots?

ga-zelle, 2022-09-03:
This is what it looked like in version 2 .8
[image]

Here is a screenshot from 3.0.0.1
[image]

And this one was a minute ago in 3.1.0.2:
[image]

By the way, in the logfile those messages are nicely separated as usual.

Andries-Smit, 2022-09-03:
Looks like semi column was replace by a new line before. Could be fixed easy. There are some semi columns missing in the debug string... Will check that too

ga-zelle, 2022-09-03:
Interesting; I never inserted newline or semcolon for my extra debug messages before.

Andries-Smit, 2022-09-03:
You are right, guess this might be handled a bit different after the conversion to Kotlin.
https://github.com/nightscout/AndroidAPS/blob/master/app/src/main/java/info/nightscout/androidaps/plugins/aps/logger/LoggerCallback.kt#L22
I expected that the next line would do the trick
``` kt
logBuffer.appendLine(obj1.toString())
```
But it did not :( 
Might @MilosKozak know what the the real fix is.

ga-zelle, 2022-09-04:
I am pretty sure it was nearly solved in 3.0.0.1, apart from the first few lines. I happen to have a screen shot from that version:
[image]

justmara, 2022-10-06:
@ga-zelle [this line](https://github.com/nightscout/AndroidAPS/blob/master/app/src/main/java/info/nightscout/androidaps/plugins/aps/openAPSSMB/OpenAPSSMBFragment.kt#L140) removes line-breaks. and it is replicated over all plugins for now already :)

ga-zelle, 2022-10-06:
Thank you @justmara for digging so deep. I 'll try dropping the Replace part, not the whole line.

olorinmaia, 2026-09-23:
This has been fixed/improved in dev (AAPS 4.0.0) closing issue

closed 2026-09-23 by olorinmaia