nightscout/AndroidAPS #2062: No correction by tab "Treatments"

Viewquest, 2022-09-07:
After the last update of 3.1.0.3-dev-a I noticed an error:
So far I have been able to correct an incorrect entry of the bolus using the "Treatments" button. Unfortunately, I can no longer activate the delete function.

Andries-Smit, 2022-09-07:
On small phones, it is hidden in the menu.
 - Click on the 3 dots on the right,
 - Click on delete
 - Select the items you like to delete, with the checkboxes
 - Click the delete button
 - Click to confirm the deletion

This was implemented in https://github.com/nightscout/AndroidAPS/pull/1362

Viewquest, 2022-09-08:
Thank you so much for your answer.  unfortunately your suggestion doesn't work.  I could do the same with the old version.  am the only one with this error?  will download the apk to my old phone today.  let's see if it works there.

Andries-Smit, 2022-09-10:
@Viewquest  I dont understand, what does work and what does not. What stap do you take an Do you got an error message?
Can you add a screenshots?

Alamo04, 2022-09-10:
> After the last update of 3.1.0.3-dev-a I noticed an error:
> So far I have been able to correct an incorrect entry of the bolus using the "Treatments" button. Unfortunately, I can no longer activate the delete function.

**Confirmed.**
Updated now to the last version, but still no **delete** in treatments possible.

@Andries-Smit 
The Trash Button in treatments has nothing to say anymore, nothing happens.
Please try it with the latest dev's.

> On small phones, it is hidden in the menu. For larger phones, a trashcan icon is shown.
> 
>     * Click on the three dots on the right,
> 
>     * Click on delete
> 
>     * Select the items you like to delete, with the checkboxes
> 
>     * Click the delete button
> 
>     * Click to confirm the deletion
> 
> 
> This was implemented in #1362

This implementation is gone and not working anymore. 
There is no error message, there will only nothing happen if you try that.
In the 3 dot menu is now only "Remove AAPS started entries" avaible and so far I can see, that has also nothing to say.

Viewquest, 2022-09-11:
Hello Andries,
My problem is that I can no longer delete using the "Treatment" button. When I click on the trash can, there is no longer a box to click like before. Nothing happens. So I can no longer make any corrections. That's why I suggested making the APK available in Dropbox. So you could see the error yourself. Am I the only one with this error?
Many greetings
Lothar

Viewquest, 2022-09-11:
there is no new DEV from Milos.  which I have now installed is very good.  the Dash works fine and XDrip and Byoda also work.

marco76, 2022-09-11:
Same problem with a build done with the Dev branch sources (version of the 10.09.22). The Bin icon has no effect. I reinstalled a Dev version with the sources of the 21.08.22 and the problem was not present (Bin icon -> select treatment -> etc. works correctly).

philipgo, 2022-09-12:
Milos fixed this with [https://github.com/nightscout/AndroidAPS/commit/0ccc5bfdfcb7f3f17f18d47594a8b3f42ca1bcd4](url)

Alamo04, 2022-09-12:
> Milos fixed this with [0ccc5bf](https://github.com/nightscout/AndroidAPS/commit/0ccc5bfdfcb7f3f17f18d47594a8b3f42ca1bcd4)

Confirmed, on my system.

Viewquest, 2022-09-13:
Hello milos,

thank you very much for your correction. The bug is fixed and can fully enjoy my 3.1.0.3 -dev-a.

Best regards
Lothar

closed 2022-09-15 by MilosKozak