nightscout/AndroidAPS #2063: XHR poll error

robertrub, 2022-09-07:
I'm testing a personal data storage on a NAS for NS. All is setup, xDrip can send BGs to it, several navigators can see the NS page. The NAS side is working.

The problem is with the AAPS side. When I setup AAPS NS settings with **https://mynasdns:4443** I systematically get _xhr poll error_ 

On the NAS side, I see that the connection is on and the data seems to arrive (at least the numbers are identical).

I don't want to attach logs as I don't know what connection information will be included...

[image]

[image]

Several people talk about this problem on Discord but no solution seems found for the moment.

CaptainBalou, 2022-09-07:
Hello @robertrub,

I had the same behaviour on my initial Nightscout setup (docker). AndroidAPS tried to load all entries in the database to Nightscout and threw those errors. The Nightscout process crashed in the same time. After restarting it was running again for a while until the error came again. After some hundreds of `xhr poll error` and the corresponding re-start of the docker container the old data was uploaded finally. Since this I have never had a crash with this error again. So I haven't investigated further.

Regards
Marco

parapenT1sta, 2022-09-07:
Are you using master or dev branch from Nightscout?

robertrub, 2022-09-07:
@parapenT1sta  I use Dev-A last version.

swissalpine, 2022-09-07:
Maybe this issue will help?
https://github.com/nightscout/AndroidAPS/issues/75

robertrub, 2022-09-10:
Nope, couldn't find any help for my situation there. I tried many things but the :port part seems to jinx AAPS (when it works in xDrip and navigators)

robertrub, 2022-09-10:
Testing different address in AAPS.

My .fly.dev address is working fine. If I add :443 to the end, it still works.

The .myqnapcloud.com:4443 address starts the connection, goes ok till the _LAST_ line, stops for several seconds than gives the xhr poll error

[image]

I have another DNS to my NAS. The dans.net address does NOT have a letsencrypt certificate. If I use the .ddns.net:4443 adress in AAPS, the first part doesn't even happen. It goes directly to the xhr error.

[image]

robertrub, 2022-09-10:
Yes, yes AND yes :) I did it. Now, I have a working AAPS connection to my QNAP NAS. The only thing to do was to select WebSocket instead of HTTP on the reverse proxy of the NAS. 

[image]

closed 2022-09-10 by robertrub

LoopLuke100, 2023-10-08:
@robertrub Hello! Can I reopen this topic? 
I have AAPS created with fly.dev, Atlas Mongo DB. The sugar levels get transferred to NS without problems but the data from AAPS to NS got the same error as yours. Unfortunately, I don't understand the Web Socket thing. Can you help me a little bit where to look for that?

robertrub, 2023-10-09:
Hi

What app is sensing BGs to NS ?

Which version of AAPS are you using ?

Can you access the HTTPS (with the S!) address with different browsers with any security messages / errors ?