nightscout/AndroidAPS #2067: [3.1.0.3] Loop stuck on basal rate despite instance of low BG. Pump unreachable issue?

nlof, 2022-09-08:
Using Omni Eros via Emalink and AAPS 3.1.0.3 (screenshot with built version)

It seems that there was a period of ~ 5 hours (from ~ 5:30 to ~ 10:30) in which loop was stuck on basal rate. Before and after there were SMB's and varying rates. Normal behavior returned after AAPS restart.

During those 5 hours, at around 8 am BG reached a sudden low of 62, which seemed to correct on its own afterwards but the basal rate didn't change one bit.

At around 9:30 a pump unreachable error was signaled (it's set to 30 minutes). Only around 10:30 I manually restarted AAPS which immediately fixed the issue and normal pump variability restarted.

I think that the pump was unreachable for longer extent of time during those 5 hours, which caused this behavior. It may have been intermittent or not sure why it was only signaled at 9:30.
A simple reboot cured this issue as it always does.

I'm not sure this is the explanation for those 5 hours of stuck loop but I frequently get pump unreachable and the simplest cure is to exit AAPS which immediately restarts and fixes the problem with the pump reconnecting right away.

As a suggestion for a new feature, maybe an AAPS reboot if "pump unreacheable" could be incorporated in AAPS as an on off toggle option (on off box or switch)?

[AndroidAPS._2022-09-08_05-11-36_.3.zip](https://github.com/nightscout/AndroidAPS/files/9525474/AndroidAPS._2022-09-08_05-11-36_.3.zip)
[AndroidAPS._2022-09-08_05-11-36_.4.zip](https://github.com/nightscout/AndroidAPS/files/9525475/AndroidAPS._2022-09-08_05-11-36_.4.zip)
[AndroidAPS._2022-09-08_05-11-36_.5.zip](https://github.com/nightscout/AndroidAPS/files/9525476/AndroidAPS._2022-09-08_05-11-36_.5.zip)
[AndroidAPS._2022-09-08_05-11-36_.6.zip](https://github.com/nightscout/AndroidAPS/files/9525477/AndroidAPS._2022-09-08_05-11-36_.6.zip)
[AndroidAPS._2022-09-08_05-11-36_.7.zip](https://github.com/nightscout/AndroidAPS/files/9525478/AndroidAPS._2022-09-08_05-11-36_.7.zip)
[AndroidAPS._2022-09-08_05-11-36_.8.zip](https://github.com/nightscout/AndroidAPS/files/9525479/AndroidAPS._2022-09-08_05-11-36_.8.zip)
[AndroidAPS._2022-09-08_05-11-36_.9.zip](https://github.com/nightscout/AndroidAPS/files/9525480/AndroidAPS._2022-09-08_05-11-36_.9.zip)
[AndroidAPS._2022-09-08_05-11-36_.27.zip](https://github.com/nightscout/AndroidAPS/files/9525481/AndroidAPS._2022-09-08_05-11-36_.27.zip)
[AndroidAPS._2022-09-08_05-11-36_.28.zip](https://github.com/nightscout/AndroidAPS/files/9525482/AndroidAPS._2022-09-08_05-11-36_.28.zip)
[AndroidAPS._2022-09-08_05-11-36_.29.zip](https://github.com/nightscout/AndroidAPS/files/9525483/AndroidAPS._2022-09-08_05-11-36_.29.zip)
[AndroidAPS._2022-09-08_05-11-36_.30.zip](https://github.com/nightscout/AndroidAPS/files/9525484/AndroidAPS._2022-09-08_05-11-36_.30.zip)
[AndroidAPS._2022-09-08_05-11-36_.31.zip](https://github.com/nightscout/AndroidAPS/files/9525485/AndroidAPS._2022-09-08_05-11-36_.31.zip)
[AndroidAPS._2022-09-08_10-08-30_.0.zip](https://github.com/nightscout/AndroidAPS/files/9525486/AndroidAPS._2022-09-08_10-08-30_.0.zip)
[AndroidAPS._2022-09-08_10-08-30_.1.zip](https://github.com/nightscout/AndroidAPS/files/9525487/AndroidAPS._2022-09-08_10-08-30_.1.zip)
[AndroidAPS._2022-09-08_10-08-30_.2.zip](https://github.com/nightscout/AndroidAPS/files/9525488/AndroidAPS._2022-09-08_10-08-30_.2.zip)
[AndroidAPS._2022-09-08_10-41-00_.0.zip](https://github.com/nightscout/AndroidAPS/files/9525489/AndroidAPS._2022-09-08_10-41-00_.0.zip)
[AndroidAPS._2022-09-08_10-41-00_.1.zip](https://github.com/nightscout/AndroidAPS/files/9525490/AndroidAPS._2022-09-08_10-41-00_.1.zip)
[AndroidAPS._2022-09-08_10-41-00_.2.zip](https://github.com/nightscout/AndroidAPS/files/9525512/AndroidAPS._2022-09-08_10-41-00_.2.zip)

[image]
[image]
[image]

nlof, 2022-09-08:
Just adding a screenshot an hour later with what normally my loop does (cuts off basal rate) as the BG goes down towards 70..
[image]

MilosKozak, 2022-09-09:
did you disable playprotect?

nlof, 2022-09-09:
Yep, it's been off since I started running AAPS. The app is running but it occasionally disconnects from the pump. The most efficient cure is to restart AAPS, it reconnects immediately.

nlof, 2022-09-09:
BTW, you can see from the screenshot that the app was running because the temp target line was changing from time to time during the fixed basal rate period.

MilosKozak, 2022-09-10:
[image]
stop exactly at 5 am is typical for playprotect

MilosKozak, 2022-09-10:
in logs is visible that AAPS really cannot connect to pump. this is android thing, nothing what could AAPS resolve itself

do you have bt watchdogs disabled in aaps and xdrip?

nlof, 2022-09-10:
It has nothing to do with playprotect. It happens randomly througout the day, rarely at night. Couple of times it happened in attached screenshots.
I do have watchdog enabled jn AAPS, maybe I'll try with it off.

Xdrip doesn't use Bluetooth in this setup because I use BYODA as collector. I do use xdrip just to keep a record but it doesn't use Bluetooth.

And Bluetooth via BYODA works obviously because I get regular BG readings throughout that time.

[image]
[image]

nlof, 2022-09-10:
I do believe it is related to AAPS because a simple restart of AAPS immediately causes a reconnection with the pump. 
Moreover, it could be a constructive fix I think to add the option of AAPS rebooting if it doesn't connect with the pump for a user specified amount of minutes. And add it as an on/off option, with the default being off, so only whomever has this kind of a problem will use it.

I might have to teach myself programming again (used to run programs in Basic on Commodore Vic 20.in the 1980's 😀) maybe to implement such a fix myself, hope it's not too complicated.

nlof, 2022-10-10:
A weird instance which may be related to the same thing, unclear.

I found the app screen stuck on an SMB 0.1u request at around 11:45AM. Restarted app and saw that the pump was stuck on basal rate since ~ 6:30AM. Uploading all the logs since. 
There was no pump unreachable warning at any time...
Luckily the basal rate handled BG great anyway.

[image]
[image]

[AndroidAPS._2022-10-10_11-38-25_.1.zip](https://github.com/nightscout/AndroidAPS/files/9744512/AndroidAPS._2022-10-10_11-38-25_.1.zip)
[AndroidAPS._2022-10-10_11-38-25_.0.zip](https://github.com/nightscout/AndroidAPS/files/9744513/AndroidAPS._2022-10-10_11-38-25_.0.zip)
[AndroidAPS._2022-10-10_00-00-03_.70.zip](https://github.com/nightscout/AndroidAPS/files/9744514/AndroidAPS._2022-10-10_00-00-03_.70.zip)
[AndroidAPS._2022-10-10_00-00-03_.69.zip](https://github.com/nightscout/AndroidAPS/files/9744515/AndroidAPS._2022-10-10_00-00-03_.69.zip)
[AndroidAPS._2022-10-10_00-00-03_.68.zip](https://github.com/nightscout/AndroidAPS/files/9744516/AndroidAPS._2022-10-10_00-00-03_.68.zip)
[AndroidAPS._2022-10-10_00-00-03_.67.zip](https://github.com/nightscout/AndroidAPS/files/9744517/AndroidAPS._2022-10-10_00-00-03_.67.zip)
[AndroidAPS._2022-10-10_00-00-03_.66.zip](https://github.com/nightscout/AndroidAPS/files/9744518/AndroidAPS._2022-10-10_00-00-03_.66.zip)
[AndroidAPS._2022-10-10_00-00-03_.65.zip](https://github.com/nightscout/AndroidAPS/files/9744519/AndroidAPS._2022-10-10_00-00-03_.65.zip)
[AndroidAPS._2022-10-10_00-00-03_.64.zip](https://github.com/nightscout/AndroidAPS/files/9744520/AndroidAPS._2022-10-10_00-00-03_.64.zip)
[AndroidAPS._2022-10-10_00-00-03_.63.zip](https://github.com/nightscout/AndroidAPS/files/9744521/AndroidAPS._2022-10-10_00-00-03_.63.zip)
[AndroidAPS._2022-10-10_00-00-03_.62.zip](https://github.com/nightscout/AndroidAPS/files/9744522/AndroidAPS._2022-10-10_00-00-03_.62.zip)
[AndroidAPS._2022-10-10_00-00-03_.61.zip](https://github.com/nightscout/AndroidAPS/files/9744523/AndroidAPS._2022-10-10_00-00-03_.61.zip)
[AndroidAPS._2022-10-10_00-00-03_.60.zip](https://github.com/nightscout/AndroidAPS/files/9744524/AndroidAPS._2022-10-10_00-00-03_.60.zip)
[AndroidAPS._2022-10-10_00-00-03_.59.zip](https://github.com/nightscout/AndroidAPS/files/9744525/AndroidAPS._2022-10-10_00-00-03_.59.zip)
[AndroidAPS._2022-10-10_00-00-03_.58.zip](https://github.com/nightscout/AndroidAPS/files/9744526/AndroidAPS._2022-10-10_00-00-03_.58.zip)
[AndroidAPS._2022-10-10_00-00-03_.57.zip](https://github.com/nightscout/AndroidAPS/files/9744527/AndroidAPS._2022-10-10_00-00-03_.57.zip)
[AndroidAPS._2022-10-10_00-00-03_.56.zip](https://github.com/nightscout/AndroidAPS/files/9744528/AndroidAPS._2022-10-10_00-00-03_.56.zip)
[AndroidAPS._2022-10-10_00-00-03_.55.zip](https://github.com/nightscout/AndroidAPS/files/9744529/AndroidAPS._2022-10-10_00-00-03_.55.zip)
[AndroidAPS._2022-10-10_00-00-03_.54.zip](https://github.com/nightscout/AndroidAPS/files/9744530/AndroidAPS._2022-10-10_00-00-03_.54.zip)
[AndroidAPS._2022-10-10_00-00-03_.53.zip](https://github.com/nightscout/AndroidAPS/files/9744531/AndroidAPS._2022-10-10_00-00-03_.53.zip)
[AndroidAPS._2022-10-10_00-00-03_.52.zip](https://github.com/nightscout/AndroidAPS/files/9744532/AndroidAPS._2022-10-10_00-00-03_.52.zip)
[AndroidAPS._2022-10-10_00-00-03_.51.zip](https://github.com/nightscout/AndroidAPS/files/9744533/AndroidAPS._2022-10-10_00-00-03_.51.zip)
[AndroidAPS._2022-10-10_00-00-03_.50.zip](https://github.com/nightscout/AndroidAPS/files/9744534/AndroidAPS._2022-10-10_00-00-03_.50.zip)
[AndroidAPS._2022-10-10_00-00-03_.49.zip](https://github.com/nightscout/AndroidAPS/files/9744535/AndroidAPS._2022-10-10_00-00-03_.49.zip)

closed 2023-04-17 by nlof