nightscout/AndroidAPS #2069: [3.1.0.3-dev-a_a995867cda] Report of 2 bugs - Treatment and Messages

Alamo04, 2022-09-11:
In the last dev versions are two new bug:

Treatments:
- deleting of wrong entry's not longer possible. Delete button and "Remove AAPS started entrys" didn't have any function now.
[image]

Messages:
- messages, that comes up if you enter wrong values (basal, isf...) shows only a white box. Think its a white background with also white letters ?
[image]

@MilosKozak 
if you are aware of this bugs, you can close this issue.

My System:
Pixel 5, DanaRS, Dexcom G6 with Boyda, latest Dev AAPS.

justmara, 2022-09-11:
Have you tried pressing this icon? It may help with deleting records ;)
[image]

philipgo, 2022-09-11:
Same bugs here on a completely different setup

MilosKozak, 2022-09-12:
menu fixed by https://github.com/nightscout/AndroidAPS/commit/0ccc5bfdfcb7f3f17f18d47594a8b3f42ca1bcd4

Alamo04, 2022-09-12:
Hi @MilosKozak,

- Treatment deletion works now. :)

- White letters on white background are still remaining in Profile (Basal and IC). ISF and DIA works correct.

MilosKozak, 2022-09-15:
@osodebailar

osodebailar, 2022-09-15:
@MilosKozak i will have a look on it the next days

closed 2022-09-15 by MilosKozak