nightscout/AndroidAPS #2080: Wrong import treatment form NS

misiek2lucky, 2022-09-16:
Hi. I found some bug.

First two pictures are from AAPS 3.1.0.3

Second two pictures are from My AAPS Client 3.1.0.3

 Third from NS

In the same time we  first add ecarbs 40g for 240min and using calc 40g and insulin for meal. AAPS Client don't import 40g from meal from NS. my preview is misleading me, and I might react wrongly.

[image]
[image]
[image]
[image]
[image]

MilosKozak, 2022-09-16:
use dev version of NS for proper sync

misiek2lucky, 2022-09-16:
Thank You I will try.

closed 2022-09-22 by MilosKozak