nightscout/AndroidAPS #2085: 3.1.0.3 DEV-B AAPS restarts and kills db

robertrub, 2022-09-21:
Installation of dev-b makes AAPS restart all the time (probably kills the database).

Going back to dev-a, AAPS starts with an empty db.

Der-Schubi, 2022-09-21:
Same here.
`09-21 09:35:23.520 19758 19758 D AndroidRuntime: Shutting down VM                                                                                                                                                                                                                                                             
09-21 09:35:23.520 19758 19758 E AndroidRuntime: FATAL EXCEPTION: main                                                                                                                                                                                                                                                        
09-21 09:35:23.520 19758 19758 E AndroidRuntime: Process: info.nightscout.androidaps, PID: 19758                                                                                                                                                                                                                              
09-21 09:35:23.520 19758 19758 E AndroidRuntime: java.lang.NoSuchMethodError: No virtual method getParcelableExtra(Ljava/lang/String;Ljava/lang/Class;)Ljava/lang/Object; in class Landroid/content/Intent; or its super classes (declaration of 'android.content.Intent' appears in /system/framework/framework.jar)         
09-21 09:35:23.520 19758 19758 E AndroidRuntime:        at info.nightscout.androidaps.receivers.BTReceiver.onReceive(BTReceiver.kt:17)                                                                                                                                                                                        
09-21 09:35:23.520 19758 19758 E AndroidRuntime:        at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1679)                                                                                                                                          
09-21 09:35:23.520 19758 19758 E AndroidRuntime:        at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(Unknown Source:2)                                                                                                                                                                      
09-21 09:35:23.520 19758 19758 E AndroidRuntime:        at android.os.Handler.handleCallback(Handler.java:938)                                                                                                                                                                                                                
09-21 09:35:23.520 19758 19758 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:99)                                                                                                                                                                                                                
09-21 09:35:23.520 19758 19758 E AndroidRuntime:        at android.os.Looper.loopOnce(Looper.java:201)                                                                                                                                                                                                                        
09-21 09:35:23.520 19758 19758 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:288)                                                                                                                                                                                                                            
09-21 09:35:23.520 19758 19758 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:7870)                                                                                                                                                                                                          
09-21 09:35:23.520 19758 19758 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)                                                                                                                                                                                                                     
09-21 09:35:23.520 19758 19758 E AndroidRuntime:        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)                                                                                                                                                                                  
09-21 09:35:23.520 19758 19758 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1003)                               `

Der-Schubi, 2022-09-21:
It's this commit from Monday: https://github.com/nightscout/AndroidAPS/commit/be30b3cd3de05db2589a5c6f5a7e3fc19544b930

app/src/main/java/info/nightscout/androidaps/receivers/BTReceiver.kt:
-        val device: BluetoothDevice = intent.getParcelableExtra(BluetoothDevice.EXTRA_DEVICE) ?: return
+        val device = intent.getParcelableExtra(BluetoothDevice.EXTRA_DEVICE, BluetoothDevice::class.java) ?: return

Der-Schubi, 2022-09-21:
I made a PR for this: https://github.com/nightscout/AndroidAPS/pull/2086

mick711, 2022-09-21:
Same problem need to go back and lost data

Der-Schubi, 2022-09-21:
PR to fix this: https://github.com/nightscout/AndroidAPS/pull/2089

closed 2022-09-21 by MilosKozak