nightscout/AndroidAPS #2090: Request to includge Blucon nightrider App as BG source

docgeo87, 2022-09-21:
Hi admin,
Kindly include blucon night rider app as primary BG source in future updates. Im from India in india there is only libre pro version which cannot pair to xdrip.Hence kindly enable blucon app in BG source for smooth flow of data into android aps.please kindly consider this humble request.
Much love ❤
Abey George
Android aps user.

MilosKozak, 2022-09-21:
I don't know it. How do you use it now?

docgeo87, 2022-09-21:
Hello MilosKozak!
Gald to hear from you...
At the moment im using libropro>blucon>NS>android aps
I see that mioamiao is being used as primary BG source in androidaps but blucon nightrider is not included in androidaps as BG source.Kindly request you to include for easy accesibility.
[image]

Andries-Smit, 2022-09-21:
@docgeo87 Did you try Bluecon to connect direct to xDrip?
https://github.com/NightscoutFoundation/xDrip/blob/38fee03be56da6ce9cbe90e08dbe8e4b651c6edc/Documentation/blukon.md
If that works for you xDrip can be used as source for AAPS

docgeo87, 2022-09-21:
Hi andries....i have been using libre pro version and there is no way to connect it with xdrip i believe this applies for libre and libre 2? Please correct me...thank you.

closed 2026-10-15 by MilosKozak