nightscout/AndroidAPS #2091: Statistics UI refinement

szantos, 2022-09-21:
We have a much better Statistics page now, as before.

[image]

Here are some of my ideas for a refinement:

The columns Bolus, Basal and Basal% make no sense with SMB, and limited sense with AMA.
While seeing TDD, Carbs and TIR values in the same line provide an informative overview, if profile change/tuning is needed.

Changes (v2):
Delete columns: Bolus, Basal, Basal%
Rename column Sum to TDD
Move column carbs to the left
Move TIR(3,9-10,0) to the right beside TDD and Carbs
Move title TIR(3,9-10,0) above 3 right columns
Delete titles: average, today, TDD, etc.
Rename Title Average (3,9-7,8) to TIR (3,9-7,8)
Move 30 days under 07 days
Separate card for today on top
Separate card for TIR (3,9-7,8) on the bottom

Notice the space freed up with dumping Bolus, Basal, Basal% and uniting the spreadsheets, too.

[image]

Changes (v3):
Move units to column titles - to let the eye focus on the values

[image]

Optional:
Coloring of values of individual days:
  red = high TDD, Carb, Below and Above and low TIR
  white = for near average values
  green = for the rest

If Bolus, Basal, Basal% seems to be very important, I'd suggest to leave only Basal%, and this could even fit in an own column in the 3rd image.

robertrub, 2022-10-05:
Love the idea 🤩 

Some doctors ask for the bolus/basal ratio though... (Totally agree that it is BS).

Not a good idea to remove units to the head of the columns. We're lost without them (need to look at the head to see what the numbers mean...). You see units, you know which column you're reading.

szantos, 2022-10-06:
I'm happy to hear that!

I'm not saying the bolus/basal ratio is BS, just that you can't tell the real ratio with these values. In my understanding the real ratio is = dailySumOfBaseBasal / TDD.

Currently:
- in case of SMB a lot of boluses contribute to base basal activity, a very low basal % is shown
- in case of AMA some higher TBRs produce activity over base basal activity, higher basal % can be shown

IMHO this is misleading.

(Ideally, the real bolus/basal ratio could be shown. In case of profil switches this can be tricky to calculate - but this would be another topic.)

kristinsmotz, 2022-10-12:
I personally prefer v2 above, I find it distracting to NOT have the designation of "what number is this" directly attached to the numbers i.e. 30U or 30%, and having to check the headings to make sure I'm reading the correct column. I'd also suggest to "rename" the TIR (Time in Range) for the 3.9-7.8 target range to TIT(Time in Target), as that corresponds better with having a "range" of normal values 3.9-10, and a "target" to aim for that's a little tighter.

dv1, 2023-01-25:
I also vote for v2.

clvsjr9, 2023-01-30:
Suggestion to permit any way to re-define the lower and upper bounds of TIR in statistics. Maybe in Config Builder --> Overview

szantos, 2023-02-07:
I've updated v2 suggstion (not in the picture):
- an additional column with the mean SGV would indicate sensitivity changes
- if today is on top, the last 7 days should decrease (newest on top)

MilosKozak, 2026-10-15:
v4 changed it

closed 2026-10-15 by MilosKozak