nightscout/AndroidAPS #2097: Dash

Viewquest, 2022-09-22:
I installed the 3.1.0.3-dev-b today and found the following errors. First I wanted to activate a new dash after deactivating the old dash. There was an Activate Pod button, but I couldn't activate it. So I had to go back to the dev-a variant. Here it worked. Please check the process and correct it if necessary. Many Thanks.
Lothar

Andries-Smit, 2022-09-22:
@Viewquest is this duplicate of #2092 ?

Viewquest, 2022-09-22:
Bi

Viewquest, 2022-09-22:
- [x] No

MilosKozak, 2022-09-22:
what was the exact verison of dev-b?

MilosKozak, 2022-09-22:
i hope activation was fixed in https://github.com/nightscout/AndroidAPS/commit/7543f3f11834bfb6a059448e6cc8fb9e11ea9bce

Viewquest, 2022-09-22:
Von meinem iPhone gesendet

> Am 22.09.2022 um 20:45 schrieb Milos Kozak ***@***.***>:
>
> what was the exact verison of dev-b?

Viewquest, 2022-09-22:
sorry, it's the same problem. Didn't see it.

Viewquest, 2022-09-22:
Flying to NY today and will fix the problem in 10 days.

Best greads
Lothar

MilosKozak, 2022-09-22:
I see, Pixel 4 .
it's already fixed

closed 2022-09-22 by MilosKozak