nightscout/AndroidAPS #2103: [3.1.0.3-dev-a] No pairing for Dana-i/RS after clicking Selected Pump

szantos, 2022-09-28:
After clicking Selected Pump in Dana-i/RS Preferences nothing happens, I'm unable to pair the pump.

- phone BT is on
- every permission for AAPS is given
- I've reset pairing information after long clicking BT status on the DANA Tab
- after short clicking BT status on the DANA Tab it tries to connect (counts the seconds)
- there were no Dana pumps paired with AAPS before (just a Combo)
- the Dana pump I'd like to pair, can communicate with the AnyDana app on the same phone

MilosKozak, 2022-09-28:
dev-a  is not the latest. test with current master

szantos, 2022-10-03:
@MilosKozak 
same with 1de236a602

MilosKozak, 2022-10-10:
tested with latest dev and no issue. scanning dialog appears immediately

MilosKozak, 2022-12-04:
can you re-test with latest dev?
provide logs pls

MilosKozak, 2022-12-09:
finally I saw the crash in logs
fixed by https://github.com/nightscout/AndroidAPS/commit/479ceabe9d064f5a7cb16d9ef93f80ff5fa9567d

closed 2022-12-09 by MilosKozak