nightscout/AndroidAPS #2110: Adding target for SMB disabling

yhonatanb20, 2022-10-02:
Hello. It would be wonderful if the target for "enable SMB with high temp targets" were customizable. Expaining: if I set a TT of 120, I want to tell aaps: "don't be too strong, I want my BG a little higher now", while if I set a TT to 180, I want tell aaps: "I'm doing sport now, so even if my BG is 170, back off, It'll go down without your help in a couple of minutes". Hence I'd love it if I could set a "disable SMB when TT is highet than ___ (ex. 140)" setting.
Thanks a lot to all developers, you made my life uncomperably better.

ga-zelle, 2022-10-07:
Similar to your idea I developed a system where any even TempTarget enables SMB and any off TempTarget disables SMB. So overnight I can use TT=91 and be safe from any SMB or during sport which needs full mental awareness I can use TT=120 allowing SMB if needed. Of course all that has to be enabled first by a dedicated switch. Otherwise or without any TT the normal AAPS rules apply.

It is only useful when using mg/dl because the forward/backward conversion is a bit funny. Would that meet your wishes?

yhonatanb20, 2022-10-07:
Not completely. What I was looking for was a more simple solution - simply adding an option to customize the threshold from which a TT will be considered "high", and therefore SMB will be disabled. Currently, any target above 100 mg/dl is considered "high" for the "enable SMB with high TT" setting.

ga-zelle, 2022-10-07:
But isn't setting TT=141 easy for disabling SMB and setting TT=142 for enabling SMB?

yhonatanb20, 2022-10-07:
It is not quite the purpose... I'd like for ex. that any TT above 140 would disable SMB

szantos, 2022-11-28:
Good idea, but new automation actions would be much more versatile:
enable/disable SMB

yhonatanb20, 2023-09-20:
Hello everybody. Have any attempts been done on this topic recently? I'll repeat the rationale: if I have a hypo and eat a small meal, I will like AAPS to give SMB to cover

pnharris02, 2025-01-22:
I'd like to suggest another potential case where being able to set the TT that cuts off SMBs would be useful.
I have had Type 1 for over 51 years now, and for a long time, I have felt symptoms of neuroglycopaenia at any blood glucose level below 5.5mmol/L (approx. 100mg/dL).  So for me, I have a normal target of 8.0mmol/L, and ideal target range of 6-10.  If I wanted to be able to use a pre-meal temp target then that would obviously be a case where SMBs should be turned on, but there is no way that I would ever set a TT below 5.5mmol/L.