nightscout/AndroidAPS #2113: minor BUG:  automation uses mmo instead of the units selected in preferences until you delete the sample.

ffsb42, 2022-10-02:
AndroiAPD 3.1.0.3 
FullRelease 

Good Afternoon,
there seems to be a minor bug with Automation.

As a new user, when you try to setup automation (obj10); there is a pre-defined example for lowBG. 
BUG1: This example uses mmo units regardless of the fact that in preferences I have selected mg/dl as units.
BUG2: even if I try to create a new automated task, the unit is still in mmo and there is no way to change it.

Workaround: 
if I delete the provided sample, I am then able to create a new automated task which properly uses mg/dl.

[image]

[image]

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2022-10-02:
rules use units selected during creation. changing UI units afterwards has no effect on existing rules

closed 2022-10-02 by MilosKozak

ffsb42, 2022-10-03:
> 

yes you are correct, the problem is that when you build the app from the source code it comes with a sample rule already made using mmo!  As a result, if you set the unit in preference to mg then walk your way through all the objectives... when you finally reach objective 10, the rule is already there still with mmo. Furthermoe, if you add new rules without deleting first the initial one, they also use mmo.

MilosKozak, 2022-10-03:
no. new rules use actual units

ffsb42, 2022-10-03:
no they don't as long as the sample rule exists. :-) 
at least they didn't for me as you can see in the screenshot above...   
you should be able to reproduce by:

1.  building the app, 
2. installing it on a brand new android
3. setup your units to mg/dl 
4. go through the objectives 1-9 (I hope you have a way to speed it up!)
5. when you start objective 10, check the values of the sample rule and confirm it shows wrong units: mmo
6. without deleting the initial sample rule, try to create a new one and confirm is shows wrong uinits: mmo
7. delete the initial rule 
8. create a new rule and confirm it now shows the correct units: mg/dl