nightscout/AndroidAPS #2114: Error Messages Omnipod Dash

Viewquest, 2022-10-04:
I installed the AAPS 3.1.0.3-dev-b on Sunday and now have the following error messages:
1. Setting temp basal might have basal failed if temp basal was previously runnig, it was canceled. pleas manually refresh pod status from the omnipod tab.
2. Unable to verify whether SMB bolus (0.35U) succeeded. <b>Refresh pod status to confirm or deny this command.
3. AndroidAPS not responding:
            close the app
            waiting
Since the messages appear more frequently, I believe that a correction should be made. Many Thanks.
Lothar
My systems: Pixel 4a, Byoda, xDrip

MilosKozak, 2022-10-04:
provide logs

Viewquest, 2022-10-04:
[AndroidAPS_LOG_1664879303140.log.zip](https://github.com/nightscout/AndroidAPS/files/9705029/AndroidAPS_LOG_1664879303140.log.zip)

Viewquest, 2022-10-09:
Hello Milos, could you do something with my logs? I look forward to an update. Many greetings
Lothar

MilosKozak, 2022-10-10:
log contains only a few minutes and connection was working fine

MilosKozak, 2022-10-10:
open new  issue and provide all info requested if you can replicate the issue
* and reset logging to default + check all PUMP*

closed 2022-10-10 by MilosKozak