nightscout/AndroidAPS #2115: Dev-b bug: loop tab, pull down, restart aaps

robertrub, 2022-10-05:
Strange bug. On the loop tab, when you pull down to force recalculation, AAPS stops or restarts. It was working OK before this beta.

Look around 0h37

[image]

[image]

robertrub, 2022-10-05:
On loop tab, 3 dots menu, _execute now_ works normally.

Only the pull down restarts AAPS.

MilosKozak, 2022-10-06:
https://github.com/nightscout/AndroidAPS/commit/d3bb0d295ecb2180356a52bd94e14bcc675b4727

closed 2022-10-06 by MilosKozak