nightscout/AndroidAPS #2118: TDD statistics scrambled

Duester64, 2022-10-07:
[image]

The lines are scrambled due to IEs are added in each line. Proposal to add IE in Head line and delete it from the lines to clean up the table.

Alamo04, 2022-10-07:
What version are you using.

Seems to be a problem with small displays.

On my Pixel 5 everything is fine, but I use the latest dev.

Duester64, 2022-10-07:
I'm using the latest 3103 dev b and I have a common Samsung S21.

MilosKozak, 2022-10-09:
you have probably large font set in system

cr582, 2022-10-10:
"large fontset" seems to be the default answer to all display problems. Maybe the developers could agree that you can make fontsets bigger for a reason and find a solution for that? But what do i know?

MilosKozak, 2022-10-10:
I agree. And it works like expected. If you you set larger font than app is designed for (or you have smaller screen), system start wrapping the text. It's android feature, not AAPS

Duester64, 2022-10-10:
Just for proposal if you reduce the  column carbohydrates an bit it would fit even for laarger fonts.

swissalpine, 2022-10-10:
It is again due to the German translation: "Kohlenhydrate" is a very long word, while the English "Carbs" is much shorter.
Maybe the translators should translate with the UI in mind and use the abbreviation "KH" instead of "carbohydrates".

szantos, 2022-10-12:
Also see Changes (v3) proposal in #2091

MilosKozak, 2022-11-30:
I added new "carbs_short" for translation

closed 2022-11-30 by MilosKozak