nightscout/AndroidAPS #2122: [3.1.0.3-dev-b] NSClient Statistics line with zeros | Force recalculate

szantos, 2022-10-12:
[image]

[image]

NSClient Statistics is sometimes unable to calculate and saves a line with zeros (and 0,1s). This compromises statistics 7 days long, until it disappears.

Reproduction:
After switching on wifi on the NSClient-phone in the morning data gets loaded and processed. If statistics are calculated during this period, it fails. Maybe it can fail other times, too.

Workaround:
Adding an ability to force recalculation?
Maybe with an additional button?

Build: c3ed8233db

MilosKozak, 2022-10-12:
adding data to that days should cause recalculation

szantos, 2022-10-12:
All data was there later that day, too. And it is still there in history browser. Waiting for history browser to fully load and opening Statistics after that, changed nothing.

MilosKozak, 2022-10-12:
add a history microbolus for example

szantos, 2022-10-12:
With NSClient there is no ability to add historical insulin

I've tried in NSClient with 1g carb:
- added for that day
- waited to show up on NS website
- checked statistics in NSClient
- "TDD calculation in progress"
- same result (with zeros)

After that 0,1U in AAPS master has been added:
- it appeared in NSClient history browser
- checked statistics in NSClient
- "TDD calculation in progress"
- same result (with zeros)

Are you sure there is a recalculation for past days, too? Results seem to be cached.

MilosKozak, 2022-10-12:
found a bug https://github.com/nightscout/AndroidAPS/commit/d6e3fa9a7510e3c6f29cb91a2dcd86c41ee2a4cb
fix is in dev

MilosKozak, 2022-11-30:
https://github.com/nightscout/AndroidAPS/commit/0ff8161cda29477d2c179332702d6d28948febf0

closed 2022-11-30 by MilosKozak