nightscout/AndroidAPS #2124: NS Client "Follower"/"Remote User" - "No profile set" warning - Not possible to activate any profile

gianlucaleuthen, 2022-10-13:
I'm using NSClient version 3.1.0.3 as a follower to my sons AAPS.

I changed my mobile (same no.) and had to re-install NSClient. After installation process the "no profile set" error came up and I can't do anything about it. I already tried to load/reset/create a local profile - no chance. Trying to activate a profile always does nothing. Additionally no treatments are possible from the mobile and all buttons for insulin, bolus, etc. on front page are missing, although set up in the settings.

My phone is Android 13 , Pixel 7 Pro, if this is of any importance/help.

Only thing that's working is the readings of current Nightscout values.

Readings from

MilosKozak, 2022-10-14:
verify AAPS can receive data from NS
do profile switch (to something different, or there and back)

Agn2020, 2022-10-23:
After reactivating (reaplying) profile on master phone NSClient have changed to master's profile.

MilosKozak, 2022-10-23:
it's related to fact that NS sends only last 2 days of data. it can/will be resolved with NSClient v3

closed 2022-10-23 by MilosKozak