nightscout/AndroidAPS #2130: [3.1.0.2] Suddenly stops invoking

PoweRGbg, 2022-10-15:
Loop is not activated on new BG. AndroidAPS is removed from all battery optimizations. Phone is Samsung S21 Ultra Android 12, One UI 4.1
--------------
- between 23:00 pm and 5:35am 
-  Build version 3.1.0.2 Bacbad75-2022.08.02
[AndroidAPS._2022-10-15_00-00-00_.6.zip](https://github.com/nightscout/AndroidAPS/files/9792057/AndroidAPS._2022-10-15_00-00-00_.6.zip)
[AndroidAPS._2022-10-15_00-00-00_.5.zip](https://github.com/nightscout/AndroidAPS/files/9792058/AndroidAPS._2022-10-15_00-00-00_.5.zip)
[AndroidAPS._2022-10-15_00-00-00_.4.zip](https://github.com/nightscout/AndroidAPS/files/9792059/AndroidAPS._2022-10-15_00-00-00_.4.zip)
[AndroidAPS._2022-10-15_00-00-00_.3.zip](https://github.com/nightscout/AndroidAPS/files/9792060/AndroidAPS._2022-10-15_00-00-00_.3.zip)
[AndroidAPS._2022-10-15_00-00-00_.2.zip](https://github.com/nightscout/AndroidAPS/files/9792061/AndroidAPS._2022-10-15_00-00-00_.2.zip)
[AndroidAPS._2022-10-15_00-00-00_.1.zip](https://github.com/nightscout/AndroidAPS/files/9792062/AndroidAPS._2022-10-15_00-00-00_.1.zip)
[AndroidAPS._2022-10-15_00-00-00_.0.zip](https://github.com/nightscout/AndroidAPS/files/9792063/AndroidAPS._2022-10-15_00-00-00_.0.zip)

yhonatanb20, 2022-10-21:
Experiencing a similiar issue on 3.1.0.3. Sometimes I "catch" the app not running the loop on a new BG reading, which can be seen on the overview screen (prediction lines extending from the previous reading rather than from the last reading) as well as on the loop screen (last run more than 5min ago):
[image]
It happens on a daily basis and I can have no indication how often, no log files generated with the same time as the issue occures.
Phone model: Samsung Galaxy A52s (Android 12), all battery optimization and play protect off
Build:
[image]
Logs:
[AndroidAPS._2022-10-21_00-00-04_.3.zip](https://github.com/nightscout/AndroidAPS/files/9838953/AndroidAPS._2022-10-21_00-00-04_.3.zip)
[AndroidAPS._2022-10-21_00-00-04_.2.zip](https://github.com/nightscout/AndroidAPS/files/9838954/AndroidAPS._2022-10-21_00-00-04_.2.zip)
[AndroidAPS._2022-10-21_00-00-04_.1.zip](https://github.com/nightscout/AndroidAPS/files/9838955/AndroidAPS._2022-10-21_00-00-04_.1.zip)
[AndroidAPS._2022-10-21_00-00-04_.0.zip](https://github.com/nightscout/AndroidAPS/files/9838956/AndroidAPS._2022-10-21_00-00-04_.0.zip)

margogoodfellow, 2022-11-06:
I have the same problem. It happen on all versions apart 3.0

yhonatanb20, 2022-11-08:
Any news on here? Tried rebooting phone each several days, helped a little bit but keep getting this issue. Very strange, but aaps doesn't invoke on each reading. It seems as if sometimes it "misses" the fact that a new reading has arrived (although it is displayed in the overview screen). Had no such issues before 3.1.0.3...

margogoodfellow, 2022-11-08:
No, no news. I have reported this a few times on here but there was always
something unclear.  Also I had this problem in every version apart or
3.0.0.1 . I have now got new build which I have completed objectives for,
new phone  new nightscout and problem has came back. I don't understand
how..

yhonatanb20, 2022-11-17:
Hey, you cannot just ignore that or blame it on the phone model! Please, what's about it, it keeps happening more and more!

MilosKozak, 2022-11-30:
@yhonatanb20 no logs according to provided screenshot

MilosKozak, 2022-11-30:
@PoweRGbg still happening with latest dev?

yhonatanb20, 2022-11-30:
I know. That's the point. There are no such logs. AAPS sometimes doesn't invoke on a new reading and doesn't realize it. All battery optimization off, rebooting and "updating" AAPS (from 3.1.0.3 to 3.1.0.3) didn't help.

PoweRGbg, 2022-12-02:
Haven't tried it yet. Will install tomorrow and give feedback here.

MilosKozak, 2022-12-04:
https://github.com/nightscout/AndroidAPS/issues/2165
try with latest dev and provide new logs.

PoweRGbg, 2022-12-08:
With the latest dev I had no such problem for the last 4 days. The database is reloading much faster which might have been causing the issue.

yhonatanb20, 2022-12-30:
The problem stopped reoccurring when updated to android 13 (One UI 5) some 3 weeks ago!!!!!!!
Probably my OS was interfering...
Thank you all for all your work :)

closed 2023-01-21 by MilosKozak