nightscout/AndroidAPS #2134: [3.1.0.3 Master branch] Cannot loop, always executing

diyaps, 2022-10-19:
Reporting bugs
--------------
[AndroidAPS.zip](https://github.com/nightscout/AndroidAPS/files/9820484/AndroidAPS.zip)
After reset my pump and phone and concent again. It  cannot loop.
The loop tap always show "Executing".

MilosKozak, 2022-10-19:
a) communication with RL is failing. That's something I cannot resolve for you. It's either phone or RL issue
b) only master and latest dev is supported here

closed 2022-10-19 by MilosKozak

diyaps, 2022-10-22:
@MilosKozak  not sure why RF is failing, but I still can bolus.  Anyway finally I resolved the issue by setting up the basal in the pump. Then it can loop.
I think some how the RF failing error make apps cannot sync profile to pump and then issue occurs