nightscout/AndroidAPS #2135: [3.1.0.3] aaps suddenly crashes at night

yhonatanb20, 2022-10-21:
"Android APS started" note appears sometimes during the night (usually around 6 a.m.). All play protect and battery optimization disabled.
Model: Samsung Galaxy A52s (Android 12)
Build:
[image]
Logs:
[AndroidAPS._2022-10-21_00-00-04_.0.zip](https://github.com/nightscout/AndroidAPS/files/9839140/AndroidAPS._2022-10-21_00-00-04_.0.zip)
[AndroidAPS._2022-10-21_00-00-04_.1.zip](https://github.com/nightscout/AndroidAPS/files/9839141/AndroidAPS._2022-10-21_00-00-04_.1.zip)
[AndroidAPS._2022-10-21_00-00-04_.2.zip](https://github.com/nightscout/AndroidAPS/files/9839142/AndroidAPS._2022-10-21_00-00-04_.2.zip)
[AndroidAPS._2022-10-21_00-00-04_.3.zip](https://github.com/nightscout/AndroidAPS/files/9839143/AndroidAPS._2022-10-21_00-00-04_.3.zip)
[AndroidAPS._2022-10-21_06-08-40_.0.zip](https://github.com/nightscout/AndroidAPS/files/9839144/AndroidAPS._2022-10-21_06-08-40_.0.zip)
[AndroidAPS._2022-10-21_06-08-40_.1.zip](https://github.com/nightscout/AndroidAPS/files/9839145/AndroidAPS._2022-10-21_06-08-40_.1.zip)
[AndroidAPS._2022-10-21_06-08-40_.2.zip](https://github.com/nightscout/AndroidAPS/files/9839147/AndroidAPS._2022-10-21_06-08-40_.2.zip)
[AndroidAPS._2022-10-21_06-08-40_.3.zip](https://github.com/nightscout/AndroidAPS/files/9839152/AndroidAPS._2022-10-21_06-08-40_.3.zip)
[AndroidAPS._2022-10-21_06-08-40_.4.zip](https://github.com/nightscout/AndroidAPS/files/9839159/AndroidAPS._2022-10-21_06-08-40_.4.zip)

MilosKozak, 2022-10-21:
I see crash around 6AM but reason is not visible from logs
I see Out of memory crashes comming from Galaxy A52s 5G but there is no identification filled so I don't know if it's you
if you are experienced user you may try dev version with many issues resolved
sorry

closed 2022-10-21 by MilosKozak

yhonatanb20, 2022-10-21:
Maybe it was me. Is there any way for me to check the reason? Would rather not use dev, a bit afraid of it having in total more issues...